openbmc/openbmc #1929: WARNING: CPU: 0 PID: 410 at /tmp/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31

PavamanSubramaniyam, 2017-07-10:
Install the latest Open BMC Firmware image on P9 Zaius Hardware.
Then in the dmesg of the ssh Open BMC console seeing the below warning messages.

```
[   84.990000] w1_master_driver w1_bus_master4: Family 0 for 00.800000000000.8c is not registered.
[   89.540000] w1_master_driver w1_bus_master3: Family 0 for 00.800000000000.8c is not registered.
[   89.540000] ------------[ cut here ]------------
[   89.540000] WARNING: CPU: 0 PID: 414 at /tmp/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x68/0x88
[   89.540000] sysfs: cannot create duplicate filename '/bus/w1/devices/00-800000000000'
[   89.540000] CPU: 0 PID: 414 Comm: w1_bus_master3 Not tainted 4.10.17-7858cdff46b6d8aa4a4f88884a0623ff50fb78c9 #1
[   89.540000] Hardware name: ASpeed SoC
[   89.540000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[   89.640000] [<8000d934>] (show_stack) from [<801dbecc>] (dump_stack+0x20/0x28)
[   89.640000] [<801dbecc>] (dump_stack) from [<80018d54>] (__warn+0xe0/0x108)
[   89.640000] [<80018d54>] (__warn) from [<80018dc0>] (warn_slowpath_fmt+0x44/0x4c)
[   89.640000] [<80018dc0>] (warn_slowpath_fmt) from [<80145d54>] (sysfs_warn_dup+0x68/0x88)
[   89.640000] [<80145d54>] (sysfs_warn_dup) from [<80146040>] (sysfs_do_create_link_sd+0x94/0xa4)
[   89.640000] [<80146040>] (sysfs_do_create_link_sd) from [<801460a4>] (sysfs_create_link+0x34/0x44)
[   89.640000] [<801460a4>] (sysfs_create_link) from [<80240868>] (bus_add_device+0x12c/0x1b8)
[   89.640000] [<80240868>] (bus_add_device) from [<8023eb0c>] (device_add+0x2a0/0x59c)
[   89.640000] [<8023eb0c>] (device_add) from [<8023ee2c>] (device_register+0x24/0x28)
[   89.640000] [<8023ee2c>] (device_register) from [<802da900>] (w1_attach_slave_device+0x1b0/0x3d4)
[   89.640000] [<802da900>] (w1_attach_slave_device) from [<802dac78>] (w1_slave_found+0xb8/0xbc)
[   89.640000] [<802dac78>] (w1_slave_found) from [<802db53c>] (w1_search+0x258/0x2b8)
[   89.640000] [<802db53c>] (w1_search) from [<802dc818>] (w1_search_devices+0x50/0x54)
[   89.640000] [<802dc818>] (w1_search_devices) from [<802db61c>] (w1_search_process_cb+0x80/0x12c)
[   89.760000] [<802db61c>] (w1_search_process_cb) from [<802db844>] (w1_process+0xf4/0x13c)
[   89.760000] [<802db844>] (w1_process) from [<800335ec>] (kthread+0x124/0x164)
[   89.760000] [<800335ec>] (kthread) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
[   89.760000] ---[ end trace f2795c7221ab5514 ]---
```

PavamanSubramaniyam, 2017-07-10:
The Firmware level information is as follows:

```
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-109"
VERSION_ID="v1.99.7-109-g143da74"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-109"
BUILD_ID="v1.99.7"

root@zaius:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x02820000..0x02821000 !
[                                                  ] 0%open-power-zaius-8937c2f-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-158-ga1e0a047b2a0
        hostboot-f576402
        linux-4.11.6-openpower1-p40b0a48
        petitboot-v1.4.3-p952b4ee
        machine-xml-46a2b3c
        occ-7adb49a
        hostboot-binaries-0e2ad15
        capp-ucode-9c73e9f
        sbe-02021c6
[==================================================] 100%
```

PavamanSubramaniyam, 2017-07-10:
[dmesg.txt](https://github.com/openbmc/openbmc/files/1135272/dmesg.txt)
[journalctl.txt](https://github.com/openbmc/openbmc/files/1135274/journalctl.txt)

GHF, 2017-07-10:
This and #975, #1790, and #1833 are duplicates. We don't see this on recent Zaius hardware even with no thermal sensor boards plugged in (the 1-wire bus pullup resistors are on the boards...).

williamspatrick, 2017-07-11:
```
Apr 20 23:05:01 zaius openpower-proc-control[2020]: Failed to open the device.
Apr 20 23:05:01 zaius systemd[1]: vcs_workaround@0.service: Main process exited, code=exited, status=255/n/a
Apr 20 23:05:02 zaius systemd[1]: Failed to start Run VCS workaround on host0.
```

This is the failure that is preventing boot.

williamspatrick, 2017-07-11:
@GHF @spinler - Is there a known issue with the VCS workaround on Zaius?

```
Apr 21 00:59:17 zaius systemd[1]: Started Scan FSI devices.
Apr 21 00:59:17 zaius systemd[1]: Starting Run VCS workaround on host0...
Apr 21 00:59:19 zaius systemd[1]: Stopped Scan FSI devices.
```

Is there a dependency wrong?  The 'scan FSI devices' and 'vcs workaround' seems like it was started almost in parallel, but it is hard to tell if the 'scan FSI devices' just happened very fast.

spinler, 2017-07-11:
The FSI step will spit out at least a few dozen lines of trace information like:
```
Jul 10 20:53:58 witherspoon systemd[1]: Starting Scan FSI devices...
Jul 10 20:53:59 witherspoon kernel:  fsi0: fsi: found chip c0010d12 at 00:00:00
Jul 10 20:53:59 witherspoon kernel:  slave@00:00: engine[2]: type 4, version 6, addr c00 size 400
```
If that isn't there, then the scan couldn't find anything for some reason.

GHF, 2017-07-11:
Oh I see. This comes down again to whether you're on EVT1 motherboards, because the current firmware only has FSI GPIO mappings for EVT2/3 and DVT. BMC FSI won't work on EVT1 hardware without tweaking the device tree entries for the GPIO mappings.

williamspatrick, 2017-07-11:
It turns out there were no processors in this system... :/

closed 2017-07-11 by williamspatrick