openbmc/openbmc #1930: [Palmetto] Host watchdog timed out

swe12345, 2017-07-10:
```
root@palmetto:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running 
```
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-109"
VERSION_ID="v1.99.7-109-g143da74"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-109"
BUILD_ID="v1.99.7"
```
```
root@palmetto:~# cat /etc/timestamp
20170710012514
```
```
root@palmetto:~# cat "/tmp/out.txt"
open-power-palmetto-v1.17-140-gadc94d6
        buildroot-2017.05-15-gcd06d25
        skiboot-v5.7-rc1
        hostboot-695bd89
        linux-4.11.6-openpower1-p8a0eff4
        petitboot-v1.4.3-pe451daa
        machine-xml-c6f5639-p908064f
        occ-28f2cec
        hostboot-binaries-0e2ad15
        capp-ucode-9c73e9f
```
```
obmc-enable-host-watchdog@0.service - Start Watchdog0
   Loaded: loaded (/lib/systemd/system/obmc-host-start@0.target.wants/../obmc-enable-host-watchdog@.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2017-07-10 11:34:28 UTC; 2min 35s ago
  Process: 1173 ExecStart=/usr/sbin/mapper call /xyz/openbmc_project/watchdog/host%i org.freedesktop.DBus.Properties Set ssv xyz.openbmc_project.State.Watchdog Enabled b true (code=exited, status=1/FAILURE)
 Main PID: 1173 (code=exited, status=1/FAILURE)

Jul 10 11:34:28 palmetto systemd[1]: Starting Start Watchdog0...
Jul 10 11:34:28 palmetto mapper[1173]: Error finding '/xyz/openbmc_project/watchdog/host0' service: No such file or directory
Jul 10 11:34:28 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Main process exited, code=exited, status=1/FAILURE
Jul 10 11:34:28 palmetto systemd[1]: Failed to start Start Watchdog0.
Jul 10 11:34:28 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Unit entered failed state.
Jul 10 11:34:28 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Failed with result 'exit-code'.
root@palmetto:~# 
```
```

1
--
Jul 10 11:34:28 palmetto systemd[1]: Starting Start Watchdog0...Jul 10 11:34:28 palmetto mapper[1173]: Error finding '/xyz/openbmc_project/watchdog/host0' service: No such file or directoryJul 10 11:34:28 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Main process exited, code=exited, status=1/FAILUREJul 10 11:34:28 palmetto systemd[1]: Failed to start Start Watchdog0.Jul 10 11:34:28 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Unit entered failed state.Jul 10 11:34:28 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Failed with result 'exit-code'

```

[journalctl_10jul.txt](https://github.com/openbmc/openbmc/files/1135324/journalctl_10jul.txt)

swe12345, 2017-07-10:
```
Entry : /xyz/openbmc_project/logging/entry/3
{
    "AdditionalData": [],
    "Id": 3,
    "Message": "org.open_power.Host.Error.WatchdogTimedOut",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Informational",
    "Timestamp": 1499686454025,
    "associations": []
}
Entry : /xyz/openbmc_project/logging/entry/2
{
    "AdditionalData": [],
    "Id": 2,
    "Message": "org.open_power.Host.Error.WatchdogTimedOut",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Informational",
    "Timestamp": 1499686404997,
    "associations": []
}
Entry : /xyz/openbmc_project/logging/entry/1
{
    "AdditionalData": [],
    "Id": 1,
    "Message": "org.open_power.Host.Error.WatchdogTimedOut",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Informational",
    "Timestamp": 1499686356708,
    "associations": []
} 
```

williamspatrick, 2017-07-10:
This indicates that the host failed to boot and failed to send progress updates (watchdog resets) to the BMC. Do you have the host console output?

gkeishin, 2017-07-10:
@williamspatrick The Host SOL didn't show any the istep sequence.  Doesn't look like the SBE sequence started at all

```
Jul 10 06:08:10 palmetto systemd[1]: Starting Init PNOR for Host0...
Jul 10 06:08:11 palmetto control_host.exe[926]: Booting host
Jul 10 06:08:11 palmetto control_host.exe[926]: ERROR HostControl: GPIO sequence failed (rc=2)
Jul 10 06:08:10 palmetto kernel[836]: [  409.100000] ftgmac100 1e660000.ethernet eth0: NCSI interface up
Jul 10 06:08:11 palmetto systemd[1]: Started Init PNOR for Host0.
Jul 10 06:08:11 palmetto systemd[1]: Started Start Host0.
Jul 10 06:08:12 palmetto phosphor-chassis-state-manager[919]: Received signal that power ON is complete
Jul 10 06:08:12 palmetto phosphor-chassis-state-manager[919]: Change to Chassis Power State
```

gkeishin, 2017-07-25:
@swe12345  Did u still see this in newer BMC and PNOR build ?

swe12345, 2017-07-27:
Observing the same with the latest driver
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-71"
VERSION_ID="v1.99.8-71-ge319c8c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-71"
BUILD_ID="v1.99.8"
```

geissonator, 2017-07-27:
Time to get someone with cronus on the system to see why sbe/host isn't starting (like Uma).

swe12345, 2017-08-01:
Driver Info:
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-121"
VERSION_ID="v1.99.8-121-gb7716b2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-121"
BUILD_ID="v1.99.8"
```
```
-bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://9.3.164.147//xyz/openbmc_project/logging/entry/2
{
  "data": {
    "AdditionalData": [],
    "Id": 2,
    "Message": "org.open_power.Host.Error.WatchdogTimedOut",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1501564394296,
    "associations": []
  },
  "message": "200 OK",
  "status": "ok"
```
```
Aug 01 05:12:58 palmetto dropbear[1071]: PAM password auth succeeded for 'root' from ::ffff:9.3.62.216:50297
Aug 01 05:12:58 palmetto dropbear[1072]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Aug 01 05:12:58 palmetto dropbear[1072]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Aug 01 05:13:13 palmetto phosphor-watchdog[1031]: Timer Expired
Aug 01 05:13:13 palmetto systemd[1]: Created slice system-openpower\x2ddebug\x2dcollector\x2dwatchdog.slice.
Aug 01 05:13:13 palmetto systemd[1]: Starting OpenPOWER debug data collector for host watchdog timeout...
Aug 01 05:13:13 palmetto watchdog_timeout[1076]: Host watchdog timed out
Aug 01 05:13:14 palmetto phosphor-fru-fault-monitor[897]: /xyz/openbmc_project/logging/entry/2
Aug 01 05:13:14 palmetto systemd[1]: Started OpenPOWER debug data collector for host watchdog timeout.
Aug 01 05:13:14 palmetto systemd[1]: Reached target Quiesce Target.
Aug 01 05:13:14 palmetto systemd[1]: Reached target Host instance 0 timed out refreshing Watchdog.
Aug 01 05:13:14 palmetto phosphor-host-state-manager[921]: Auto reboot enabled. Reducing HOST BOOTCOUNT by 1.
Aug 01 05:13:14 palmetto sensor_manager2.py[838]: Setting boot count to 2
Aug 01 05:13:14 palmetto phosphor-host-state-manager[921]: Beginning reboot...
Aug 01 05:13:14 palmetto phosphor-host-state-manager[921]: Host State transaction request
```

swe12345, 2017-08-02:
Observing the same with the latest driver
Driver Info
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-147"
VERSION_ID="v1.99.8-147-g9a33a15"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-147"
BUILD_ID="v1.99.8"
root@palmetto:~#
```
```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://xx.xx.xx.xx/xyz/openbmc_project/logging/entry/enumerate
{
  "data": {
    "/xyz/openbmc_project/logging/entry/2": {
      "AdditionalData": [],
      "Id": 2,
      "Message": "org.open_power.Host.Error.WatchdogTimedOut",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1501662658101,
      "associations": []
    },
    "/xyz/openbmc_project/logging/entry/3": {
      "AdditionalData": [],
      "Id": 3,
      "Message": "org.open_power.Host.Error.WatchdogTimedOut",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1501662706358,
      "associations": []
    }
  },
  "message": "200 OK",
  "status": "ok"
}-bash-4.1$
```

sivassrr, 2017-08-03:
Aug 03 05:58:32 palmetto systemd[1]: Reached target Fan Control Ready.
Aug 03 05:58:33 palmetto control_host.exe[904]: Booting host
Aug 03 05:58:33 palmetto systemd[1]: Started Start Host0.
Aug 03 05:58:33 palmetto systemd[1]: Reached target Host0 (Starting).
Aug 03 05:58:33 palmetto systemd[1]: Reached target Host0 (Started).
Aug 03 05:58:33 palmetto systemd[1]: Starting Start Watchdog0...
Aug 03 05:58:34 palmetto mapper[1262]: Error finding '/xyz/openbmc_project/watchdog/host0' service: No such file or directory   <====
Aug 03 05:58:34 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Main process exited, code=exited, status=1/FAILURE
Aug 03 05:58:34 palmetto systemd[1]: Failed to start Start Watchdog0.
Aug 03 05:58:34 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Unit entered failed state.
Aug 03 05:58:34 palmetto systemd[1]: obmc-enable-host-watchdog@0.service: Failed with result 'exit-code'.
Aug 03 05:58:34 palmetto systemd[1]: Started PCIE Slot Presence Detection For Chassis0.
Aug 03 05:58:35 palmetto systemd[1]: Reached target Start Host0.
Aug 03 05:58:35 palmetto phosphor-host-state-manager[905]: Received signal that host is running
Aug 03 05:58:35 palmetto phosphor-host-state-manager[905]: Change to Host State
Aug 03 05:58:35 palmetto control_host.exe[904]: Using primary side of the bios flash
Aug 03 05:58:35 palmetto system_manager.py[868]: Running System State: HOST_BOOTING

sivassrr, 2017-08-03:
Palmetto with following level, Able to boot all the way to Petitboot. 
root@palmetto:~# /usr/sbin/pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x01ff7000..0x01ff8000 !
[                                                  ] 0%open-power-palmetto-922a1c1
	buildroot-211bd05
	skiboot-5.3.6
	hostboot-3ed4df7
	linux-4.4.21-openpower1-d43e553
	petitboot-v1.3.0-d1ee745
	palmetto-xml-c6f5639-d76d973
	occ-bb092e9
	hostboot-binaries-7f593a3
	capp-ucode-1bb7503
[==================================================] 100%
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-60"
VERSION_ID="v1.99.8-60-g4403eb0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-60"
BUILD_ID="v1.99.8"
root@palmetto:~# cat /etc/timestamp
20170720162457
root@palmetto:~#

 Petitboot (v1.3.0-d1ee745)                         TN71-BP012 0000000000000000
 ──────────────────────────────────────────────────────────────────────────────

  System information
  System configuration
  Language
  Rescan devices
  Retrieve config from URL
 *Exit to shell

sivassrr, 2017-08-03:
With Palmetto Failure system, Looks like we are hitting PNOR corruption. 
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-147"
VERSION_ID="v1.99.8-147-g9a33a15"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-147"
BUILD_ID="v1.99.8"
root@palmetto:~# cat /etc/timestamp
20170801192517
root@palmetto:~# /usr/sbin/pflash -r /dev/stdout -P VERSION
FFS: Flash header not found. Code: 100
Error 100 opening ffs !
root@palmetto:~# /usr/sbin/pflash -r /dev/stdout -P VERSION
FFS: Flash header not found. Code: 100
Error 100 opening ffs !
root@palmetto:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running

@swe12345 could you try PDU recycle.  and try to power on.  

Also try to do following:

1) Do new BMC codeupdate with PNOR code update with XYZ
2) Then do new PNOR code update using XYZ .

swe12345, 2017-08-03:
Driver Info:
```
root@palmetto:/tmp# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-178"
VERSION_ID="v1.99.8-178-g7a75570"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-178"
BUILD_ID="v1.99.8"
```
System reached Quiesced State
```
root@palmetto:/tmp# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
root@palmetto:/tmp#
```
```
root@palmetto:/tmp# cat /etc/timestamp
20170803072531
```
```
root@palmetto:/tmp# cat "/tmp/out.txt"
open-power-palmetto-v1.17-159-gfd58b15
        buildroot-2017.05-93-g9b91216
        skiboot-v5.7-rc1
        hostboot-695bd89
        linux-4.12-openpower1-p265e6a6
        petitboot-v1.4.3-pe451daa
        machine-xml-c6f5639-p908064f
        occ-28f2cec
        hostboot-binaries-858e352
        capp-ucode-9c73e9f
```
Booted till istep 20
```
 19.38455|  STRING
 19.38455|  Hostboot Build ID:
 19.38456|================================================
 20.17448|================================================
 20.17448|Error reported by hwas (0x0C00)
 20.17448|  Insufficient hardware to continue.
 20.17449|  ModuleId   0x03 MOD_CHECK_MIN_HW
 20.17449|  ReasonCode 0x0c04 RC_SYSAVAIL_INSUFFICIENT_HW
 20.18425|  UserData1   : 0x0000000000000000
 20.18426|  UserData2   : 0x0000000000000000
 20.18426|User Data Section 0, type UD
 20.18426|  Subsection type 0x15
 20.18426|  ComponentId hb-trace (0x3100)
 20.18426|User Data Section 1, type UD
 20.18427|  Subsection type 0x06
 20.18427|  ComponentId errl (0x0100)
 20.18427|  CALLOUT
 20.18427|  PROCEDURE ERROR
 20.18427|  Procedure: 1
 20.18428|User Data Section 2, type UD
 20.18428|  Subsection type 0x01
 20.18428|  ComponentId errl (0x0100)
 20.18428|  STRING
 20.18428|  libistepdisp.so
 20.18429|User Data Section 3, type UD
 20.18429|  Subsection type 0x03
 20.18429|  ComponentId errl (0x0100)
 20.18429|User Data Section 4, type UD
 20.18430|  Subsection type 0x01
 20.18430|  ComponentId errl (0x0100)
 20.18430|  STRING
 20.18430|  Hostboot Build ID:
 20.18430|================================================
[... 29 lines cut ...]
Aug 03 09:51:34 palmetto phosphor-fru-fault-monitor[934]: Invalid Inventory Path.
Aug 03 09:51:34 palmetto phosphor-fru-fault-monitor[934]: /xyz/openbmc_project/logging/entry/96
Aug 03 09:53:13 palmetto phosphor-watchdog[1494]: Timer Expired
Aug 03 09:53:14 palmetto systemd[1]: Starting OpenPOWER debug data collector for host watchdog timeout...
Aug 03 09:53:14 palmetto watchdog_timeout[1535]: Host watchdog timed out
Aug 03 09:53:15 palmetto phosphor-fru-fault-monitor[934]: /xyz/openbmc_project/logging/entry/97
Aug 03 09:53:15 palmetto systemd[1]: Started OpenPOWER debug data collector for host watchdog timeout.
Aug 03 09:53:15 palmetto systemd[1]: Reached target Quiesce Target.
Aug 03 09:53:15 palmetto systemd[1]: Reached target Host instance 0 timed out refreshing Watchdog.
```
```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://9.3.164.147/xyz/openbmc_project/logging/entry/97
{
  "data": {
    "AdditionalData": [],
    "Id": 97,
    "Message": "org.open_power.Host.Error.WatchdogTimedOut",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1501753995152,
    "associations": []
  },
  "message": "200 OK",
  "status": "ok"
```

closed 2019-04-27 by stale[bot]