openbmc/openbmc #1931: IPMI FRU : non-host inventory

dkodihal, 2017-07-10:
#1559 and #1560 make it possible to map host inventory to IPMI FRUs. This story is to map non-host (BMC inventory) to IPMI FRUs.

devenrao, 2017-07-13:
- [ ] Create config file for non host inventory items

- [ ] Add native recipe that consumes non host inventory config file and generates fru config file by executing gen_ipmi_fru perl script
fd

- [ ] Generate fru details by consuming non host config yaml file and append to existing C header file

- [ ] Make changes to IPMI fru read code to cater for non host inventory items

geissonator, 2017-07-13:
Need story in 2.1 backlog for using MRW, for this story we'll hand code file for witherspoon.

rfrandse, 2017-08-09:
<https://gerrit.openbmc-project.xyz/5542> openpower - cater for generating FRU map for non host FRU's
Resolves  #1931 IPMI FRU : non-host inventory

closed 2017-08-30 by williamspatrick