openbmc/openbmc #1932: REST Upload - does not return any output

anoo1, 2017-07-10:
Previously the `/upload/image` would return "200 OK". But in latest master now it returns no output:

```
# curl -c cjar -b cjar -k -H "Content-Type: application/octet-stream" -X PUT -T /tmp/gurevich/witherspoon.pnor.squashfs.tar https://$BMCIP/upload/image/
#
```

gkeishin, 2017-07-11:
Even for POST operation on upload the JSON response content is missing.

```
Start / End / Elapsed:     20170711 11:27:50.195 / 20170711 11:27:50.370 / 00:00:00.175
11:27:50.369     INFO     Post Request using : alias=openbmc, uri=/upload/image, data=,                     headers={u'Content-Type': u'application/octet-stream', u'Accept': u'application/octet-stream'}, files=None, allow_redirects=True     
11:27:50.369     INFO     ${resp} = <Response [200]>
````

Looks like there is a fix from Brad.. we need to try there out

closed 2017-07-11 by williamspatrick

gkeishin, 2017-07-11:
Came clean
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.7"
VERSION_ID="ibm-v1.99.7-0-r2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.7"
BUILD_ID="ibm-v1.99.7-0-r2"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Wed Jun 28 16:42:22 CDT 2017 armv6l GNU/Linux
root@witherspoon:~#

REST Get Message JSON Format Compliance Test :: Verify REST messag... | PASS |
------------------------------------------------------------------------------
REST Post Message JSON Format Compliance Test :: Verify REST messa... ....URI:https://xx.xx.xx.xx/xyz/openbmc_project/state/host0//attr/CurrentHostState, method:Get, args:{}
Response code:200, Content:{
  "data": "xyz.openbmc_project.State.Host.HostState.Off",
  "message": "200 OK",
  "status": "ok"
}
TEST_HISTORY:                                     /gsa/ausgsa-p10/03/indiateam04/gkeishin/openbmc-test-automation/logs//test_history.txt
REST Post Message JSON Format Compliance Test :: Verify REST messa... | PASS |
------------------------------------------------------------------------------
REST Put Message JSON Format Compliance Test :: Verify REST messag... .URI:https://xx.xx.xx.xx/xyz/openbmc_project/state/host0//attr/RequestedHostTransition, method:Put, args:{u'data': {u'data': 'xyz.openbmc_project.State.Host.Transition.Off'}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
......URI:https://9xx.xx.xx.xx/xyz/openbmc_project/state/host0//attr/CurrentHostState, method:Get, args:{}
Response code:200, Content:{
  "data": "xyz.openbmc_project.State.Host.HostState.Off",
  "message": "200 OK",
  "status": "ok"
}
TEST_HISTORY:                                     /gsa/ausgsa-p10/03/indiateam04/gkeishin/openbmc-test-automation/logs//test_history.txt
REST Put Message JSON Format Compliance Test :: Verify REST messag... | PASS |
------------------------------------------------------------------------------
Check Response Codes HTTP_UNSUPPORTED_MEDIA_TYPE :: REST "Post" re... ..URI:https://xx.xx.xx.xx/xyz/openbmc_project/state/host0//attr/CurrentHostState, method:Get, args:{}
Response code:200, Content:{
  "data": "xyz.openbmc_project.State.Host.HostState.Off",
  "message": "200 OK",
  "status": "ok"
}
TEST_HISTORY:                                     /gsa/ausgsa-p10/03/indiateam04/gkeishin/openbmc-test-automation/logs//test_history.txt
Check Response Codes HTTP_UNSUPPORTED_MEDIA_TYPE :: REST "Post" re... | PASS |
------------------------------------------------------------------------------
```