openbmc/openbmc #1933: [xCAT] Powering off Chassis doesn't reflect correct state in Host (and vice versa?)

whowutwut, 2017-07-10:
Power off of the Chassis with hard off doesn't reflect in the current host0 state .  I hvaen't tested the other way around, if we send OFF to the Host, does it reflect in the Chassis state? 

```
  "data": {
    "/xyz/openbmc_project/state/bmc0": {
      "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
      "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
      "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.Off",
      "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
      "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
      "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Reboot"
    }
  },```

Is this by design?  

Which Power State should we go off of if we want to return the power status of the OS side?  If they do not sync up, then it will be difficult to provide the status, we need to hold information about which piece got the off command, host0 or chassis0... 

What is the real difference between chassis and host? 
 * Chassis power off doesn't shut power to the BMC, it's still running

geissonator, 2017-07-16:
I've got a variety of defects and issues in this area.  My goal with this issue is going to be ensure the use has a way to perform a "hard power off".  A power off of the system that will not wait for the host.  The idea was this would be via the chassis object but due to systemd and the way we've setup the targets, this doesn't quite work.

- [ ] Draw out state diagram of current on and off targets (try and put in format that can be uploaded to docs)
- [ ] Propose solution to ensure an immediate power off can occur, and can also be done after a soft power off has been requested (but is waiting for the host)
- [ ] Implement solution

**# STATUS**
7/16 - No progress on this one yet, focused on last sprints story (code review comments)

whowutwut, 2017-07-19:
Adding this to cross reference to xcat project... This fix is required to resolve the issue reported here:  https://github.com/xcat2/xcat-core/issues/3498

whowutwut, 2017-07-20:
@geissonator  monitoring the status here since this issue has affected some of our other test teams using xCAT.  I put on 1730B firmware and the softoff timeout is still at 45 minutes, that makes this a little more difficult to work around..  I see you have a plan from yesterday, is there an ETA? 

Looking at the driver schedule and the 1732 driver cutoff is 07/21 for the 08/02 driver... , if we miss that, then the next official driver build it would avail 08/16 (1734A/B), almost a month away..

geissonator, 2017-07-20:
We'll definitely cherry pick this one into  an official driver as soon as it's available @whowutwut (and not wait for the normal release cadence).

rfrandse, 2017-07-21:
<https://gerrit.openbmc-project.xyz/5425> Use the new hard power off target for chassis off requests
Resolves  #1933 [xCAT] Powering off Chassis doesn't reflect correct state in Host (and vice versa?)

closed 2017-07-24 by williamspatrick