openbmc/openbmc #1934: [xCAT] Unable to understand how to turn on LED lights on and off against the group based on the documentation.

whowutwut, 2017-07-10:
Using firmware 1728B: `ibm-v1.99.7-0-r2` 

@vishwabmc  I've tried to go through the README found here: [LED README](https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/Led) and I'm unable to successfully post the Assert against the group for LED. 

### The GET is fine...

```
[root@fs3 mid05tor12cn11]# IP=172.20.140.111
[root@fs3 mid05tor12cn11]# curl -c cjar -k -X POST -H 'Content-Type: application/json' -d '{"data": [ "root", "0penBmc" ] }' https://$IP/login
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}[root@fs3 mid05tor12cn11]# curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://$IP/xyz/openbmc_project/led/groups/enclosure_identiy
{
  "data": {
    "Asserted": 0
  },
  "message": "200 OK",
  "status": "ok"
```

### Trying to PUT

```
[root@fs3 mid05tor12cn11]# cat enclosure_led_on.json
{
  "data": {
    "Asserted": 1
  },
}
[root@fs3 mid05tor12cn11]# curl -c cjar -b cjar -k -H "Content-Type: application/json" -X PUT -T enclosure_led_on.json https://$IP/xyz/openbmc_project/led/groups/enclosure_identify
{
  "data": {
    "description": "Internal Server Error",
    "exception": "AttributeError(\"Attribute 'parameter_list' not defined.\",)",
    "traceback": [
      "Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle",
      "    return route.call(**args)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 721, in __call__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 778, in wrap",
      "    return callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 956, in __call__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 930, in wrap",
      "    return self.to_jsonp(callback(*a, **kw))",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper",
      "    rv = callback(*a, **ka)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 895, in wrap",
      "    data = callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 183, in __call__",
      "    return getattr(self, 'do_' + request.method.lower())(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 501, in do_put",
      "    req = set(request.parameter_list.keys())",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1394, in __getattr__",
      "    raise AttributeError('Attribute %r not defined.' % name)",
      "AttributeError: Attribute 'parameter_list' not defined."
    ]
  },
  "message": "500 Internal Server Error",
  "status": "error"
```

* I've tried this with XML 
* I've tried with `-d '{"data": [ "Asserted", "1" ] }'`
* I've tried with ` -d '{"data": [ "Asserted", "True" ] }'`

The https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Led/Group.interface.yaml Lists options as "True" and "False".  The README file uses "true", is this case insensitive?  The data returned has "Assert: 0" , so can we pass in and combination of true/True/1 as valid values?

vishwabmc, 2017-07-12:
I talked to @whowutwut and gave the commands to assert the group.

whowutwut, 2017-07-12:
Thanks @vishwabmc, I overlooked the PUT doc in the rest-api.md but I see now we have to pass in attr/<name> into the REST path.     Sorry about that.  

I am able to send the command, however, no LED change or lights up on the box. 
```
[root@fs3 mid05tor12cn11]# ./identify_on.sh
++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data":1}' https://172.20.140.111//xyz/openbmc_project/led/groups/enclosure_identify/attr/Asserted
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data":1}' https://172.20.140.111//xyz/openbmc_project/led/groups/fan0_identify/attr/Asserted
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data":1}' https://172.20.140.111//xyz/openbmc_project/led/groups/fan1_identify/attr/Asserted
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data":1}' https://172.20.140.111//xyz/openbmc_project/led/groups/fan2_identify/attr/Asserted
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}++ curl -b cjar -k -H Content-Type:application/json -X PUT -d '{"data":1}' https://172.20.140.111//xyz/openbmc_project/led/groups/fan3_identify/attr/Asserted
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
```

Is this expected to turn on/off LEDs?

whowutwut, 2017-07-12:
Fan LEDs do not light up in the front and NO front LED lights up for Identify.. . but an LED  DOES light up in the back.  This is good enough for xCAT's beacon light... so I am OK to close this issue... 

@bradbishop  Do you want to use this to track why other LEDs are not lighting up?  feel free to close if not.

whowutwut, 2017-07-12:
@spinler  thanks for the info. Will close this since I've addressed the initial issue

closed 2017-07-12 by whowutwut