openbmc/openbmc #1936: BMC: Determine how systemd handles the boot sequence on a read-only volume

anoo1, 2017-07-10:


williamspatrick, 2017-07-21:
I can boot up a Romulus image in QEMU with a RO rootfs and no RW file systems with pretty minimal failures.  `clear-once` is an old initramfs thing we probably won't need anymore.  `dropbearkey` is expected due to non-writable /etc.  `systemd-hostnamed` was already failing.  `systemd-tmpfiles-setup` needs some investigation.

```
root@romulus:~# systemctl --no-pager | grep failed
● clear-once.service                                                                                                loaded failed failed    Clear one time boot overrides                                                                  
● dropbearkey.service                                                                                               loaded failed failed    SSH Key Generation                                                                             
● systemd-hostnamed.service                                                                                         loaded failed failed    Hostname Service                                                                               
● systemd-tmpfiles-setup.service                                                                                    loaded failed failed    Create Volatile Files and Directories                                                          
```

```
root@romulus:~# systemctl status --no-pager systemd-tmpfiles-setup.service
● systemd-tmpfiles-setup.service - Create Volatile Files and Directories
   Loaded: loaded (/lib/systemd/system/systemd-tmpfiles-setup.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-07-21 21:49:05 UTC; 23min ago
     Docs: man:tmpfiles.d(5)
           man:systemd-tmpfiles(8)
  Process: 763 ExecStart=/bin/systemd-tmpfiles --create --remove --boot --exclude-prefix=/dev (code=exited, status=1/FAILURE)
 Main PID: 763 (code=exited, status=1/FAILURE)

Jul 21 21:49:05 romulus systemd-tmpfiles[763]: symlink(../usr/share/zoneinfo/UTC, /etc/localtime) failed: Read-only file system
Jul 21 21:49:05 romulus systemd-tmpfiles[763]: symlink(../proc/self/mounts, /etc/mtab) failed: Read-only file system
Jul 21 21:49:05 romulus systemd-tmpfiles[763]: /srv does not exist and cannot be created as the file system is read-only.
Jul 21 21:49:05 romulus systemd-tmpfiles[763]: "/var/tmp" already exists and is not a directory.
Jul 21 21:49:05 romulus systemd-tmpfiles[763]: "/var/log" already exists and is not a directory.
Jul 21 21:49:05 romulus systemd-tmpfiles[763]: /var/cache does not exist and cannot be created as the file system is read-only.
Jul 21 21:49:05 romulus systemd[1]: systemd-tmpfiles-setup.service: Main process exited, code=exited, status=1/FAILURE
Jul 21 21:49:05 romulus systemd[1]: Failed to start Create Volatile Files and Directories.
Jul 21 21:49:05 romulus systemd[1]: systemd-tmpfiles-setup.service: Unit entered failed state.
Jul 21 21:49:05 romulus systemd[1]: systemd-tmpfiles-setup.service: Failed with result 'exit-code'.
```

rfrandse, 2017-08-25:
done

closed 2017-08-25 by rfrandse