openbmc/openbmc #1937: BMC: Split u-boot into 2 volume sectors

anoo1, 2017-07-10:


mdmillerii, 2017-07-11:
Define env as two (redundant) environment copies each one 64k nor sector so one will always be valid even when erasing and writing the other.  For dual flash systems each flash will have a u-boot and pair of environment regions.

geissonator, 2017-07-13:
Need task list, Eddie to talk with Milton (hopefully mostly a uboot config change)

eddiejames, 2017-07-13:
- [ ] Modify u-boot AST config for redundancy.
- [ ] Define redundant chunks in the fw_env config.
- [ ] Make sure everything works.

eddiejames, 2017-07-13:
## Status: July 13, 2017
Discovered how to edit u-boot redundancy config, and enabled it. Tested this on Barreleye with a variety of settings. Still need to test on Witherspoon and make sure shutdown script can be safely modified.

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5312> Add u-boot redundant environments.
Resolves  #1937 BMC: Split u-boot env into 2 copies

closed 2017-08-18 by williamspatrick