openbmc/openbmc #1940: BMC: Ability for u-boot to execute the kernel that's in a ubi volume

anoo1, 2017-07-10:


maksymko, 2017-07-10:
Do you want the kernel in a volume separate from User Space (what is now rwfs)?

mdmillerii, 2017-07-11:
Yes, the plan is each fit is in it's own volume, probably named with as type-hash like pnor in Gerrit, and the primary and alternate image names will be in u-boot environment variables and boot scripts will be templates using these variables.  Similar variables tell the kernel the mtds for ubi and the ubi volume with root file system.

Current plan is only var and etc will be rw and do not need versions (Just reset & factory reset).

anoo1, 2017-07-13:
- [ ] Determine the boot env variables needed for u-boot to find the kernel in a ubi volume.
- [ ] Have the bmc updater script update the boot env variables with the kernel volume information when activation is successful.

rfrandse, 2017-08-10:
<https://gerrit.openbmc-project.xyz/5911> uboot: Add mtd support and env variables to boot kernel and rootfs
Resolves  #1940 BMC: Ability for u-boot to execute the kernel that's in a ubi volume

closed 2017-08-19 by williamspatrick