openbmc/openbmc #1942: BMC: Change primary and alt BMC layout to just have u-boot, boot env, and a ubi mtd

anoo1, 2017-07-10:


maksymko, 2017-07-10:
I think U-Boot env can also be in UBI partition.

mdmillerii, 2017-07-11:
It can read its environment from an ubi volume, but support is read only and only one ubi container can be accessed, which would make try there times then fall back to a known image difficult or impossible from env and alternate boot images on different flash devices with separate ubi space difficult.

eddiejames, 2017-07-12:
- [ ] Modify partition definition in device tree.
- [ ] Enable new partitions for Witherspoon only
- [ ] Create a means of selecting whether or not to use the new partitions.

eddiejames, 2017-07-12:
07/12: Modified partition definitions. Did some research on uboot and investigated how the OpenBMC filesystem works as a whole.

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5316> kernel: Patch in bmc flash partition changes
Resolves  #1942 BMC: Change primary and alt BMC layout to just have u-boot, boot env, and a ubi mtd

closed 2017-08-29 by williamspatrick