openbmc/openbmc #1943: BMC: Tell u-boot where squashfs image is

anoo1, 2017-07-10:


anoo1, 2017-07-13:
Status:
07/13 - New function in the bmc code update script to set the bootargs variable to the name of the volume where the squashfs image is.

anoo1, 2017-08-03:
With a kernel with changes from #1935:

```
bootargs=console=ttyS4,115200n8 ubi.mtd=pnor,0,0,6 ubi.block=6,6 ro rootfstype=squashfs root=/dev/ubiblock6_6

# df -T
Filesystem           Type       1K-blocks      Used Available Use% Mounted on
/dev/root            squashfs       14336     14336         0 100% /

# mount
/dev/ubiblock6_6 on / type squashfs (ro,relatime)
```

rfrandse, 2017-08-10:
<https://gerrit.openbmc-project.xyz/5911> uboot: Add mtd support and env variables to boot kernel and rootfs
Resolves  #1943 BMC: Tell u-boot where squashfs image is

closed 2017-08-19 by williamspatrick