openbmc/openbmc #1944: BMC: Block BMC reboot during u-boot update

anoo1, 2017-07-10:


saqibkh, 2017-07-13:
- [ ]  Create service files that can disable BMC transtition through reboot, shutdown, halt etc.
- [ ]  Call Enable reboot halt services when ActivationBlockTransition object is created.
- [ ]  Call Disable reboot halt services inside ActivationBlockTransition destructor.

Status:
May 11: 
- Discussed possible implementations with Andrew, Adriana and Patrick.
- Created service files and add added the logic to ActivationBlocksTransition class inside 
  phosphor-bmc-code-mgmt.

geissonator, 2017-07-13:
Needs more investigation

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5266> BMC: Block BMC reboor during code update
Resolves  #1944 BMC: Block BMC reboot during u-boot update

rfrandse, 2017-09-07:
<https://gerrit.openbmc-project.xyz/5266> BMC: Block BMC reboot during code update
Resolves  #1944 BMC: Block BMC reboot during u-boot update

rfrandse, 2017-09-20:
<https://gerrit.openbmc-project.xyz/6613> BMC: Block BMC reboots during Code Update
Resolves  #1944 BMC: Block BMC reboot during u-boot update

geissonator, 2018-01-03:
Lets get yocto updated to 2.4, and see if the new systemd provides the function we need.

closed 2018-02-27 by geissonator