openbmc/openbmc #1945: Netboot: Random missing dbus objects of hwmon devices

msbarth, 2017-07-10:
When netbooting an image, different hwmon devices fail to have corresponding dbus objects created even though the hwmon sysfs directories do exist. This does not seem to occur when booting from a flashed image. Checking the devices listed under hwmon sysfs directories compared to port 3000 usually shows the difference in what may be missing.

[journalctl.txt](https://github.com/openbmc/openbmc/files/1137151/journalctl.txt)
[dmesg.txt](https://github.com/openbmc/openbmc/files/1137152/dmesg.txt)

msbarth, 2017-07-10:
In my case, after re-netbooting the same image, the ambient temperature sensor which had hwmon sysfs entry(hwmon9) did have associated dbus objects.

yadron, 2017-07-13:
I've just seen (yesterday) that DBUS objects of sensors in 1.99.6.2 of Barreleye sometimes disappears too (e.g. all sensors could be read from /sys/class/hwmon/* manually without problems, but one or few of corresponding DBUS objects of sensors not found if using Rest API or any other DBUS request). 

Complete reboot (AC-power off/on) helped, all sensors back.
I've checked twice that there were no hardware errors, so this is software bug inside BMC/DBUS.

closed 2019-10-27 by stale[bot]