openbmc/openbmc #1946: Witherspoon: Jenkins Code update fails

yadlapati, 2017-07-10:
Seeing code update fails on systems running GHE code levels.

Update using force code update link works

Output from the run:
19:50:03 Started by user Miranda Pacheco
19:50:03 [EnvInject] - Loading node environment variables.
19:50:03 Building remotely on tulpok01p4-ppc64le (docker ppc64le-docker) in workspace /tmp/fr58p01-slave/workspace/OpenBMC/automated-test/helper-jobs/code-update
19:50:03 Set build name.
19:50:03 New build name is '#334-w84'
19:50:03 Next nodes: []
19:50:03 [code-update] $ /bin/bash -xe /tmp/hudson1190326328239947082.sh
19:50:03 + set -uo pipefail
19:50:03 + DOCKER_IMG_NAME=openbmc/ubuntu-obmc-code-update
19:50:03 + WORKSPACE=/tmp/fr58p01-slave/workspace/OpenBMC/automated-test/helper-jobs/code-update
19:50:03 + mkdir -p /tmp/fr58p01-slave/workspace/OpenBMC/automated-test/helper-jobs/code-update
19:50:03 + cd /tmp/fr58p01-slave/workspace/OpenBMC/automated-test/helper-jobs/code-update
19:50:03 + rm -Rf ./openbmc-build-scripts
19:50:03 + rm -Rf ./openbmc-test-automation
19:50:03 + git clone https://gerrit.openbmc-project.xyz/openbmc/openbmc-build-scripts
19:50:03 Cloning into 'openbmc-build-scripts'...
19:50:05 + cd openbmc-build-scripts
19:50:05 + . ./scripts/build-qemu-robot-docker.sh
19:50:05 ++ set -uo pipefail
19:50:05 ++ DOCKER_IMG_NAME=openbmc/ubuntu-robot-qemu
19:50:05 +++ uname -m
19:50:05 ++ '[' ppc64le == ppc64le ']'
19:50:05 ++ DOCKER_BASE=ppc64le/
19:50:05 +++ cat
19:50:05 ++ Dockerfile='FROM ppc64le/ubuntu:latest
19:50:05 
19:50:05 ENV DEBIAN_FRONTEND noninteractive
19:50:05 
19:50:05 RUN apt-get update && apt-get install -yy     debianutils     gawk     git     python     python-dev     python-setuptools     python3     python3-dev     python3-setuptools     socat     texinfo     wget     gcc     libffi-dev     libssl-dev     xterm     mwm     ssh     vim     iputils-ping     sudo     cpio     unzip     diffstat     expect     curl     build-essential
19:50:05 
19:50:05 RUN easy_install     tox     pip     requests
19:50:05 
19:50:05 RUN pip install     json2yaml     robotframework     robotframework-requests     robotframework-sshlibrary     robotframework-scplibrary
19:50:05 
19:50:05 RUN wget https://sourceforge.net/projects/ipmitool/files/ipmitool/1.8.18/ipmitool-1.8.18.tar.bz2
19:50:05 RUN tar xvfj ipmitool-*.tar.bz2
19:50:05 RUN ./ipmitool-1.8.18/configure
19:50:05 RUN make
19:50:05 RUN make install
19:50:05 
19:50:05 RUN grep -q 116 /etc/group || groupadd -g 116 fspcibld
19:50:05 RUN grep -q 57457 /etc/passwd || useradd -d /home/fspcibld -m -u 57457 -g 116                     fspcibld
19:50:05 USER fspcibld
19:50:05 ENV HOME /home/fspcibld
19:50:05 RUN /bin/bash'
19:50:05 ++ docker build -t openbmc/ubuntu-robot-qemu -
19:50:05 Sending build context to Docker daemon 3.072 kB

19:50:05 Step 1 : FROM ppc64le/ubuntu:latest
19:50:05  ---> d2e29d5dd835
19:50:05 Step 2 : ENV DEBIAN_FRONTEND noninteractive
19:50:05  ---> Using cache
19:50:05  ---> 45bb72dc486a
19:50:05 Step 3 : RUN apt-get update && apt-get install -yy     debianutils     gawk     git     python     python-dev     python-setuptools     python3     python3-dev     python3-setuptools     socat     texinfo     wget     gcc     libffi-dev     libssl-dev     xterm     mwm     ssh     vim     iputils-ping     sudo     cpio     unzip     diffstat     expect     curl     build-essential
19:50:05  ---> Using cache
19:50:05  ---> c4d3d4ad4cca
19:50:05 Step 4 : RUN easy_install     tox     pip     requests
19:50:05  ---> Using cache
19:50:05  ---> 2938b5ace4d7
19:50:05 Step 5 : RUN pip install     json2yaml     robotframework     robotframework-requests     robotframework-sshlibrary     robotframework-scplibrary
19:50:05  ---> Using cache
19:50:05  ---> 19aa2845f3b1
19:50:05 Step 6 : RUN wget https://sourceforge.net/projects/ipmitool/files/ipmitool/1.8.18/ipmitool-1.8.18.tar.bz2
19:50:05  ---> Using cache
19:50:05  ---> c99180012c23
[... 98 lines cut ...]
19:53:09 1 test total, 0 passed, 1 failed
19:53:09 ==============================================================================
19:53:09 Output:  /home/fspcibld/output.xml
19:53:10 Log:     /home/fspcibld/log.html
19:53:10 Report:  /home/fspcibld/report.html
19:53:10 Build step 'Execute shell' marked build as failure
19:53:10 Set build name.
19:53:10 New build name is '#334-w84'
19:53:10 Variable with name 'BUILD_DISPLAY_NAME' already exists, current value: '#334-w84', new value: '#334-w84'
19:53:11 [PostBuildScript] - Execution post build scripts.
19:53:11 [code-update] $ /bin/bash -xe /tmp/hudson2146762907998730904.sh
19:53:11 + curl -s http://tulpok01p4.aus.stglabs.ibm.com:8080/jenkins/job/OpenBMC/job/automated-test/job/helper-jobs/job/code-update/334//consoleText
19:53:11 Robot results publisher started...
19:53:11 -Parsing output xml:
19:53:11 Done!
19:53:11 -Copying log files to build dir:
19:53:11 Done!
19:53:11 -Assigning results to build:
19:53:11 Done!
19:53:11 -Checking thresholds:
19:53:11 Done!
19:53:11 Done publishing Robot results.
19:53:11 Collecting metadata...
19:53:11 Metadata collection done.
19:53:11 Finished: FAILURE

geissonator, 2017-07-11:
We're having REST login issues again after the prepareForUpdate reboot.  And then it started working for FFDC collection on the failure :(
```
Jul 10 13:05:40 witherspoon phosphor-gevent[891]: Traceback (most recent call last):
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     result = self._run(*self.args, **self.kwargs)
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     ciphers=ciphers)
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     self.do_handshake()
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     return self._sslobj.do_handshake()
Jul 10 13:05:40 witherspoon phosphor-gevent[891]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Jul 10 13:05:40 witherspoon phosphor-gevent[891]: <Greenlet at 0x7658b2b0: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x765677d0 fileno=5 address=('::2cc2:8047:b04d:af76%1991200160', 443, 0, 1991200160)>>(<socket at 0x7643c210 fileno=9>, ('::6059:ad76:8e91:f402%1199549204', 49184, 0, 119)> failed with SSLEOFError
Jul 10 13:05:40 witherspoon phosphor-gevent[891]: Traceback (most recent call last):
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     result = self._run(*self.args, **self.kwargs)
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     ciphers=ciphers)
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     self.do_handshake()
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Jul 10 13:05:40 witherspoon phosphor-gevent[891]:     return self._sslobj.do_handshake()
Jul 10 13:05:40 witherspoon phosphor-gevent[891]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Jul 10 13:05:40 witherspoon phosphor-gevent[891]: <Greenlet at 0x7658b580: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x765677d0 fileno=5 address=('::2cc2:8047:b04d:af76%1991200160', 443, 0, 1991200160)>>(<socket at 0x7643c1b0 fileno=10>, ('::6059:ad76:8e91:f402%1199549204', 49186, 0, 119)> failed with SSLEOFError

Jul 10 13:05:55 witherspoon phosphor-gevent[891]: ::6059:ad76:8e91:f402%1199549204 - - [2017-07-10 13:05:55] "POST /login HTTP/1.1" 200 320 0.048444
Jul 10 13:05:55 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:57666).
Jul 10 13:05:55 witherspoon dropbear[994]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jul 10 13:05:55 witherspoon dropbear[994]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jul 10 13:05:55 witherspoon dropbear[994]: Child connection from ::ffff:9.41.165.103:57666
Jul 10 13:05:56 witherspoon dropbear[994]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:57666
Jul 10 13:05:56 witherspoon dropbear[995]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Jul 10 13:05:56 witherspoon dropbear[995]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Jul 10 13:05:57 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:57668).
Jul 10 13:05:57 witherspoon dropbear[999]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jul 10 13:05:57 witherspoon dropbear[999]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jul 10 13:05:57 witherspoon dropbear[999]: Child connection from ::ffff:9.41.165.103:57668
Jul 10 13:05:58 witherspoon dropbear[999]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:57668
Jul 10 13:05:58 witherspoon dropbear[1000]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Jul 10 13:05:58 witherspoon dropbear[1000]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Jul 10 13:05:59 witherspoon phosphor-gevent[891]: ::6059:ad76:8e91:f402%1199549204 - - [2017-07-10 13:05:59] "POST /login HTTP/1.1" 200 320 0.033287
Jul 10 13:05:59 witherspoon phosphor-gevent[891]: ::6059:ad76:8e91:f402%1199549204 - - [2017-07-10 13:05:59] "GET /xyz/openbmc_project/state/BMC0/ HTTP/1.1" 404 308 0.030837
[... cut ...]
Jul 10 13:06:10 witherspoon dropbear[1025]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jul 10 13:06:10 witherspoon dropbear[1025]: Child connection from ::ffff:9.41.165.103:57690
Jul 10 13:06:11 witherspoon dropbear[1025]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:57690
Jul 10 13:06:11 witherspoon dropbear[1026]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Jul 10 13:06:11 witherspoon dropbear[1026]: lastlog_openseek: /var/log/lastlog is not a file or directory!
```

gkeishin, 2017-07-11:
This is SCP timedout issue where the network dropped and the tar image is not copied and it tried to update with invalid image 
```
19:53:09 'Unpack Error' does not contain 'to apply'
```

geissonator, 2017-07-12:
@gkeishin has kindly posted a workaround in the automated test framework for now - https://gerrit.openbmc-project.xyz/#/c/5212/

This all changes once the official code update process gets in place from @anoo1 and team so don't want to spend much time debugging the scp partial copy issue (I assume some sort of network problem).

closed 2017-07-14 by williamspatrick