openbmc/openbmc #1947: ipmitool chassis power off does not work

shenki, 2017-07-11:
Running v1.99.7-74-gec3c51e (ec3c51e) on Witherspoon, I attempted to use ipmitool to soft-power off the system:
```
ipmitool -H spoon1-bmc -I lanplus -P 0penBmc chassis power soft
```
On the BMC:
```
Jul 11 04:22:36 witherspoon systemd[1]: Started Network IPMI daemon.
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering handler: /usr/lib/net-ipmid/libapphandler.so.0
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x36]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1836
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0xFF]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command18ff
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x22]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1822
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x24]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1824
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x1]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1801
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x4]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1804
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x8]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1808
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x6]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1806
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x40]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1840
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x42]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1842
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x4], Cmd:[0xFF]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command10ff
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x4], Cmd:[0x2F]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command102f
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x4], Cmd:[0x30]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1030
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x4], Cmd:[0x2D]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command102d
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xA], Cmd:[0xFF]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command28ff
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xA], Cmd:[0x48]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command2848
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xA], Cmd:[0x49]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command2849
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xA], Cmd:[0x40]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command2840
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xA], Cmd:[0x42]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command2842
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xA], Cmd:[0x44]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command2844
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x0], Cmd:[0xFF]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Commandff
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x0], Cmd:[0x0]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command0
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x0], Cmd:[0x9]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command9
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x0], Cmd:[0x1]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x0], Cmd:[0x2]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command2
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x0], Cmd:[0x8]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command8
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x2C], Cmd:[0x3]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Commandb003
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xC], Cmd:[0xFF]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command30ff
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xC], Cmd:[0x1]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command3001
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0xC], Cmd:[0x2]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command3002
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x2]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1802
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x6], Cmd:[0x3]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1803
Jul 11 04:22:37 witherspoon netipmid[2436]: Registering NetFn:[0x2C], Cmd:[0x0]
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Commandb000
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command100000
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command120000
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command140000
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1838
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command183b
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command183c
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command10000
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1848
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command1849
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command184a
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command3021
Jul 11 04:22:37 witherspoon netipmid[2436]: I> Registering Command3022
Jul 11 04:22:37 witherspoon netipmid[2436]: >> GetChannelCapabilities
Jul 11 04:22:37 witherspoon netipmid[2436]: << GetChannelCapabilities
Jul 11 04:22:37 witherspoon netipmid[2436]: >> openSession
Jul 11 04:22:37 witherspoon netipmid[2436]: << openSession
Jul 11 04:22:37 witherspoon netipmid[2436]: >> RAKP12
Jul 11 04:22:37 witherspoon netipmid[2436]: << RAKP12
Jul 11 04:22:37 witherspoon netipmid[2436]: >> RAKP34
Jul 11 04:22:37 witherspoon netipmid[2436]: << RAKP34
Jul 11 04:22:37 witherspoon netipmid[2436]: >> setSessionPrivilegeLevel
Jul 11 04:22:37 witherspoon netipmid[2436]: << setSessionPrivilegeLevel
Jul 11 04:22:37 witherspoon netipmid[2436]: E> Table:: Command Not found: 0xb03e
Jul 11 04:22:37 witherspoon netipmid[2436]: Invalid Chassis Control command:[0x5] received
````

shenki, 2017-07-11:
It appears that soft off is not implemented.

I tried using 'chassis power off', which should perform the soft off action. That tells me:

```
Jul 11 05:22:10 witherspoon ipmid[1048]: Failed to set property in SoftPowerOff object: No route to host
Jul 11 05:22:10 witherspoon ipmid[1048]: Did not find soft off service so request Host:Transition:Off
```

vishwabmc, 2017-07-11:
`ipmitool -H spoon1-bmc -I lanplus -P 0penBmc chassis power soft`

We don't have `command:5` implemented in IPMI daemon. Need to use the SoftPowerOff dbus object to do so -or- call chassis power off ( which you seem to have done after )

yongli3, 2017-07-11:
@vishwabmc I received the same message "Failed to set property in SoftPowerOff object: No route to host" with power off command, it is normal or there is something missing on my system?

vishwabmc, 2017-07-11:
What level of code do you run ?. cat /etc/os-release ?

jk-ozlabs, 2017-07-14:
@vishwabmc : above, you say:

> Need to use the SoftPowerOff dbus object to do so **-or- call chassis power off** ( which you seem to have done after )

`chassis power off` (ie, command = 0x0) cannot be interpreted as a soft power off, this won't work with the correct power down sequence.

geissonator, 2017-07-27:
Now that I have the hard power off function working, Vishwa is going to help me out and connect it up in ipmi.

vishwabmc, 2017-07-28:
# **STATUS**

- 7/28:
    - Read the story and understood. Nothing else done on this now. Will start Monday.

rfrandse, 2017-08-04:
<https://gerrit.openbmc-project.xyz/5781> Support Chassis Soft Power Off command
Fixes  #1947 ipmitool chassis power off does not work

closed 2017-08-14 by williamspatrick