openbmc/openbmc #1949: Failed to start systemd-hostnamed.service [ Witherspoon ]

gkeishin, 2017-07-12:
Driver info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-2"
VERSION_ID="v1.99.8-2-g73a56d5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-2"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-7858cdff46b6d8aa4a4f88884a0623ff50fb78c9 #1 Wed Jul 12 15:28:52 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

```
root@witherspoon:~# systemctl status systemd-hostnamed.service | cat
● systemd-hostnamed.service - Hostname Service
   Loaded: loaded (/lib/systemd/system/systemd-hostnamed.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-04-20 17:29:47 UTC; 7min ago
     Docs: man:systemd-hostnamed.service(8)
           man:hostname(5)
           man:machine-info(5)
           http://www.freedesktop.org/wiki/Software/systemd/hostnamed
  Process: 937 ExecStart=/lib/systemd/systemd-hostnamed (code=exited, status=225/NETWORK)
 Main PID: 937 (code=exited, status=225/NETWORK)

Apr 20 17:29:45 witherspoon systemd[1]: Starting Hostname Service...
Apr 20 17:29:47 witherspoon systemd[1]: systemd-hostnamed.service: Main process exited, code=exited, status=225/NETWORK
Apr 20 17:29:47 witherspoon systemd[1]: Failed to start Hostname Service.
Apr 20 17:29:47 witherspoon systemd[1]: systemd-hostnamed.service: Unit entered failed state.
Apr 20 17:29:47 witherspoon systemd[1]: systemd-hostnamed.service: Failed with result 'exit-code'.
root@witherspoon:~#
```

Journald snippet:
```
Apr 20 17:29:57 witherspoon systemd[1]: Started Phosphor Network Manager.
Apr 20 17:30:03 witherspoon phosphor-network-manager[905]: Failed to get hostname
Apr 20 17:30:03 witherspoon phosphor-network-manager[905]: The operation failed internally.
Apr 20 17:30:03 witherspoon phosphor-network-manager[905]: Error in mapper call

Apr 20 17:29:45 witherspoon systemd[937]: systemd-hostnamed.service: Failed at step NETWORK spawning /lib/systemd/systemd-hostnamed: Invalid argument
Apr 20 17:29:47 witherspoon systemd[1]: systemd-hostnamed.service: Main process exited, code=exited, status=225/NETWORK
Apr 20 17:29:47 witherspoon systemd[1]: systemd-hostnamed.service: Unit entered failed state.
Apr 20 17:29:47 witherspoon systemd[1]: systemd-hostnamed.service: Failed with result 'exit-code'.
```
- [x] Journald 
[Host_service_startup_failure.txt](https://github.com/openbmc/openbmc/files/1142661/Host_service_startup_failure.txt)

ratagupt, 2017-07-13:
As the kernel is not configured with namespace support,and systemd looks for network namespace..I missed that patch.
I will add it.

gkeishin, 2017-07-13:
Thanks @ratagupt

ratagupt, 2017-07-13:
workaround is:-
 1) to comment the following line in the "/lib/systemd/system/systemd-hostnamed.service"
```[Unit]
Description=Hostname Service
Documentation=man:systemd-hostnamed.service(8) man:hostname(5) man:machine-info(5)
Documentation=http://www.freedesktop.org/wiki/Software/systemd/hostnamed

[Service]
ExecStart=/lib/systemd/systemd-hostnamed
BusName=org.freedesktop.hostname1
WatchdogSec=3min
CapabilityBoundingSet=CAP_SYS_ADMIN
PrivateTmp=yes
PrivateDevices=yes
#PrivateNetwork=yes >>>>>>>>>>>>>>>>comment this line.
ProtectSystem=yes
ProtectHome=yes
ProtectControlGroups=yes
ProtectKernelTunables=yes
MemoryDenyWriteExecute=yes
RestrictRealtime=yes
RestrictAddressFamilies=AF_UNIX
SystemCallFilter=~@clock @cpu-emulation @debug @keyring @module @mount @obsolete @raw-io
```
2) systemctl daemon-reload
3) systemctl restart xyz.openbmc_project.Network.service.

prkatti1, 2017-07-13:
@sivassrr @gkeishin Even our xyz network test cases failing because of this.
But works on workaround patch given by @ratagupt

rfrandse, 2017-07-20:
<https://gerrit.openbmc-project.xyz/5398> Don't return the error if unable to create the network namespace
Resolves  #1949 Failed to start systemd-hostnamed.service [ Witherspoon ]

ratagupt, 2017-07-21:
**STATUS**
 
7/20  Peer review pending
7/21 Peer review pending

closed 2017-07-25 by williamspatrick

gkeishin, 2017-07-27:
Verified working on latest build

bobarshad, 2018-06-12:
For those that need this, follow the steps below:

1. systemctl edit systemd-hostnamed

   Add the 2 lines below then exit the editor (don't forget to save when prompted):

   [Service]
   PrivateNetwork=no

2. This will create an override.conf file with the above 2 lines in the directory:

   /etc/systemd/system/systemd-hostnamed.service.d/

3. The update systemd:

   systemctl daemon-reload

4. Then restart the service:

   systemctl restart systemd-hostnamed

You should now be able to run hostnamectl without it hanging.

asgh, 2018-08-15:
The above suggestions do work, but don't do that!

Instead enable CONFIG_NET_NS - if you don't you will have other strange errors, for example pulseaudio won't work (will freeze on startup), and errors like `rtkit-daemon.service: Failed at step NETWORK spawning /usr/lib/rtkit/rtkit-daemon: Invalid argument`