openbmc/openbmc #1950: phosphor-fan-presence-tach: Failed to invoke DBus method.

gkeishin, 2017-07-12:
Driver info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-2"
VERSION_ID="v1.99.8-2-g73a56d5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-2"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-7858cdff46b6d8aa4a4f88884a0623ff50fb78c9 #1 Wed Jul 12 15:28:52 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

Journald snippet:

It seems like that message only happened 3 times, and then it was back to normal

```
Apr 20 17:31:33 witherspoon phosphor-fan-presence-tach[1230]: Failed to invoke DBus method.
Apr 20 17:31:33 witherspoon phosphor-fan-presence-tach[1230]: The operation failed internally.
Apr 20 17:31:33 witherspoon phosphor-fan-presence-tach[1230]: xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Apr 20 17:31:34 witherspoon phosphor-fan-presence-tach[1230]: Failed to invoke DBus method.
Apr 20 17:31:34 witherspoon phosphor-fan-presence-tach[1230]: The operation failed internally.
Apr 20 17:31:34 witherspoon phosphor-fan-presence-tach[1230]: xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Apr 20 17:31:34 witherspoon phosphor-fan-presence-tach[1230]: Failed to invoke DBus method.
Apr 20 17:31:34 witherspoon phosphor-fan-presence-tach[1230]: The operation failed internally.
Apr 20 17:31:35 witherspoon phosphor-fan-presence-tach[1230]: xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Apr 20 17:31:36 witherspoon phosphor-fan-presence-tach[1230]: Failed to invoke DBus method.
Apr 20 17:31:36 witherspoon phosphor-fan-presence-tach[1230]: The operation failed internally.
Apr 20 17:31:36 witherspoon phosphor-fan-presence-tach[1230]: xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
root@witherspoon:~#

```

- [x] Jounald 
[Fan_tach_journald.txt](https://github.com/openbmc/openbmc/files/1142796/Fan_tach_journald.txt)

gkeishin, 2017-07-12:
Probably this is related to above sensor error.
```
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1493]: Usage: /usr/sbin/phosphor-hwmon-readd [options]
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1493]: Options:
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1493]:     --help               print this menu
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1493]:     --path=<path>        sysfs location to monitor
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1493]:     --of-name=<path>     open firmware device name to monitor
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1493]: ERROR: Path not specified or invalid.
Apr 20 17:34:09 witherspoon systemd[1]: Started Phosphor poweron watchdog.
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1494]: Usage: /usr/sbin/phosphor-hwmon-readd [options]
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1494]: Options:
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1494]:     --help               print this menu
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1494]:     --path=<path>        sysfs location to monitor
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1494]:     --of-name=<path>     open firmware device name to monitor
Apr 20 17:34:11 witherspoon phosphor-hwmon-readd[1494]: ERROR: Path not specified or invalid.
```

gkeishin, 2017-07-12:
@bradbishop @spinler  Some help over here with info.. if we need to ignore this tach error. Thanks

spinler, 2017-07-27:
This particular problem was fixed by a fix from Brad that went in a while ago.  Closing.

closed 2017-07-27 by spinler