openbmc/openbmc #1951: Get occ sensors up and running on Zaius

geissonator, 2017-07-12:
Goal is to get similar function we have on witherspoon up and running on zaius

- bind the occ driver once booted
- update hwmon conf files to provide sensors
- enable occ pass through (amester)

geissonator, 2017-07-13:
- [x] Create OCC hwmon conf files for Zaius - https://gerrit.openbmc-project.xyz/#/c/5258/
- [x] Add openpower-occ-control application to zaius - https://gerrit.openbmc-project.xyz/#/c/5259/

STATUS:
7/13: Worked with Eddie to get needed kernel patch out, wrote and tested new hwmon conf files on top of that kernel patch (thermal sensors looking good).  Hit some roadblocks on power sensors and dynamic occ binding features, needs team discussion.

rfrandse, 2017-07-21:
<https://gerrit.openbmc-project.xyz/5259> Add occ control application into zaius
Resolves  #1951 Get occ sensors up and running on Zaius

closed 2017-07-25 by williamspatrick