openbmc/openbmc #1952: [xCAT] require a way to reset the BMC network configuration, wipe out back to DHCP

whowutwut, 2017-07-12:
Looking at issue #1575 , I've modified and created some files in `/etc/systemd/network` and I'm trying to undo any of these changes and put the BMC back to a factory DHCP state. 

I have been playing around with VLAN and created some additional files in `/etc/systemd/network`. 

```
root@witherspoon:/etc/systemd/network# ls -ltr
total 2
-rw-r--r--    1 root     root            48 Jul 12 06:09 eth0.553.netdev
-rw-r--r--    1 root     root            79 Jul 12 06:15 eth0.553.network
-rw-r--r--    1 root     root            43 Jul 12 06:15 00-bmc-eth0.network
```

But I want to reset to DHCP and lose any configuration that I have....
so removing all the files there and restarting... network...
```
root@witherspoon:/etc/systemd/network# rm *
root@witherspoon:/etc/systemd/network# ls -ltr
total 0
root@witherspoon:/etc/systemd/network#  systemctl restart systemd-networkd
```

This does remove all the configuration and put the interfaces back to DHCP, however, the `eth0.553` device still exists...

```
root@witherspoon:/etc/systemd/network# ifconfig
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet addr:172.20.253.115  Bcast:172.20.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:171542 errors:0 dropped:0 overruns:0 frame:0
          TX packets:171216 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:11240931 (10.7 MiB)  TX bytes:11538859 (11.0 MiB)
          Interrupt:20

eth0.553  Link encap:Ethernet  HWaddr 70:E2:84:14:28:0D
          inet6 addr: fe80::72e2:84ff:fe14:280d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:784 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:261063 (254.9 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:806679 errors:0 dropped:0 overruns:0 frame:0
          TX packets:806679 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:61311880 (58.4 MiB)  TX bytes:61311880 (58.4 MiB)
```

Even when I do something like...
1) remove all the files 
2) ifconfig eth0.553 down 

the eth0.553 is no longer active and seen....  but doing a `systemctl restart systemd-networkd` brings this eth0.553 device back up . 

So something else must be saving this information where we can't reset and get rid of the network configurations that were brought up.

mdmillerii, 2017-07-12:
systemd networkd is lazy, it will leave any interface configuration alone that is not explicitly mentioned.

also just a down will not remove an interface, you can see this with ifconfig -a to show all interfaces.

The ip command may be useful to try to remove interfaces, or a full reboot.

whowutwut, 2017-07-13:
@mdmillerii  Is there an equivalent of running the following commands against the OpenPower LE boxes ... (AMI)  

```
ipmitool -H ${BMC_IP} -U ${BMC_USER} -P ${BMC_PASS} raw 0x32 0xBA 00 00
ipmitool -H ${BMC_IP} -U ${BMC_USER} -P ${BMC_PASS} raw 0x32 0x66
```

Maybe ...
1) clear the `/etc/systemd/network/*` files
2) reboot the bmc

ratagupt, 2017-07-13:
@whowutwut :There is network factory reset option..Which will vipe out all the network configuration and configures the interface mode to DHCP.

ratagupt, 2017-07-13:
that option lies in the xyz.openbmc_project.Network namespace.

mtritz, 2017-07-13:
Is this an issue? Should network factory reset be able to remove extra interfaces, as in this case, or is it good as-is?

ratagupt, 2017-07-14:
Till now,I don't consider it is a defect.
@whowutwut :- Can you please close this defect?

whowutwut, 2017-07-14:
@ratagupt  No, I'm not able to get to the network endpoint... 
You keep referencing that API which I understand it's part if.. but it's not opened in the firmware. 

Which firmware level do we need to be at?  PNOR and BMC? 

This is what I ran it on.... 
```
[root@fs2vm110 p9euh02]# rinv p9euh[01,02] firm
[OpenBMC development support] Using this version of xCAT, ensure firware level is at v1.99.6-0-r1, or higher.
p9euh01: BMC Firmware Product: v1.99.7-13-gb261ea (Active)
p9euh02: BMC Firmware Product: v1.99.7-118-g5d2bf49 (Active)
```

```
[root@fs2vm110 p9euh02]# ./list.sh
+ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://10.6.9.254/xyz/openbmc_project/network/enumerate
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/network"
  },
  "message": "404 Not Found",
  "status": "error"
}[root@fs2vm110 p9euh02]# cd ../p9euh01
[root@fs2vm110 p9euh01]# ./list.sh
+ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://10.6.7.254/xyz/openbmc_project/network/enumerate
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/network"
  },
  "message": "404 Not Found",
  "status": "error"
```

whowutwut, 2017-07-14:
Listing everything under openbmc_project and greping for "net"  produced nothing...

```
[root@fs2vm110 p9euh01]#  curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://10.6.9.254/xyz/openbmc_project/list |grep net
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   111  100   111    0     0    139      0 --:--:-- --:--:-- --:--:--   139
[root@fs2vm110 p9euh01]#
```

whowutwut, 2017-07-17:
@ratagupt  Just to confirm...We can run this command on the BMC to reset? 
```
busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network \ 
xyz.openbmc_project.Common.FactoryReset Reset 
```

whowutwut, 2017-07-20:
@ratagupt  The command above that you provided didn't work... 

```
root@witherspoon:~# busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network xyz.openbmc_project.Common.FactoryReset Reset
The name xyz.openbmc_project.Network was not provided by any .service files
```

mtritz, 2017-07-24:
I just tried this myself on Witherspoon with no problems. The network reset interface is provided by xyz.openbmc_project.Network.service - perhaps you need to start or enable that?

```
root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.network
[Match]
Name=eth0
[Network]
Address=9.3.21.44/23
Gateway=9.3.20.1

root@witherspoon:/etc/systemd/network# busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network xyz.openbmc_project.Common.FactoryReset Reset

root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.network
[Match]
Name=eth0
[Network]
DHCP=true
[DHCP]
ClientIdentifier=mac

ratagupt, 2017-07-25:
@whowutwut : Can you verify it again as it is working for me and Michael?

whowutwut, 2017-07-25:
@mtritz @ratagupt  Which openbmc firmware does this need to be on? 

I know that @williamspatrick noticed yesterday that the official LAB released drivers are not even picking up 1.99.8 yet....   we are still on the .7 release... 

```
root@witherspoon:/etc/systemd/network# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-110"
VERSION_ID="v1.99.7-110-g828dd0e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-110"
BUILD_ID="v1.99.7"
```

I don't even see that service available to start....  
```
root@witherspoon:/etc/systemd/network# systemctl list-unit-files --type=service | grep Network
org.openbmc.NetworkManager.service                     enabled
```

ratagupt, 2017-08-04:
@whowutwut :- Are you still facing this issue?
If not please close this defect,Can you try with latest firmware and close this defect?

whowutwut, 2017-08-10:
@ratagupt What's the endpoint we should be calling to reset to the BMC to DHCP?

whowutwut, 2017-08-10:
@ratagupt  I can't find anything of value in the yaml files... with this search...
https://github.com/openbmc/phosphor-dbus-interfaces/search?utf8=%E2%9C%93&q=DHCP&type=

ratagupt, 2017-08-11:
https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Common/FactoryReset.interface.yaml

ratagupt, 2017-08-11:
Service:- xyz.openbmc_project.Network 
Object:- /xyz/openbmc_project/network 
Interface:- xyz.openbmc_project.Common.FactoryReset
Method:- Reset

busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network xyz.openbmc_project.Common.FactoryReset Reset

Curl command for this.

zet809, 2017-08-11:
Hi, @ratagupt , from the interface above, it seems to me that it will have the OpenBMC reset to factory. Besides reset OpenBMC to get ip through DHCP, is there any other configuration is reset?
And the REBOOT means to reboot the OpenBMC, right? What will happen if the host is running? It will be rebooted automatically? Thx!

whowutwut, 2017-08-11:
@ratagupt Thanks, let me give this a try, right now our machine that has a BMC serial connection is down with the latest PNOR, need to recover this and then I will try out that rest call

whowutwut, 2017-08-11:
@zet809 We already have the reboot function in xCAT, added here...  https://github.com/xcat2/xcat-core/pull/3438

whowutwut, 2017-08-12:
@ratagupt So with the latest firmware, I'm able to reset the code to DHCP, but ran into some issues..  but since the problem for this issue is resolved, I'll close it and open another one...

closed 2017-08-12 by whowutwut

ratagupt, 2017-08-14:
@zet809 :- Yes reboot means rebooting the openbmc.