openbmc/openbmc #1956: Gateway address on system and rest does not match

prkatti1, 2017-07-13:
Default gateway address on rest interface is not coming up properly, it always shows 0.0.0.0.

/xyz/openbmc_project/network/eth0/ipv4/25aa2ff7

    org.freedesktop.DBus.Peer
    org.freedesktop.DBus.Introspectable
    org.freedesktop.DBus.Properties
    xyz.openbmc_project.Object.Delete
    xyz.openbmc_project.Network.IP
        Address = xx.xx.xx.xx
        Gateway = 0.0.0.0
        Origin = xyz.openbmc_project.Network.IP.AddressOrigin.Static
        PrefixLength = 24
        Type = xyz.openbmc_project.Network.IP.Protocol.IPv4

ratagupt, 2017-07-14:
As dicsussed the gateway which is being shown under ip address object is not the default gateway.
The default gateway would be under /xyz/openbmc_project/network/config object.
Please confirm and close this defect.

prkatti1, 2017-07-14:
Yes, I am able to see default GW under /xyz/openbmc_project/network/config object. We can close this.

closed 2017-07-14 by prkatti1