openbmc/openbmc #1957: DCMI: Design app for data sampling/telemetry.

dkodihal, 2017-07-13:
@tomjoseph83 to add details.

tomjoseph83, 2017-08-09:
# **STATUS**

8/8
I did some background study on the requirements. Working on preparing a draft to be presented in the design meeting to solicit feedback.

tomjoseph83, 2017-09-15:
Summary of the design discussion:-

We are planning to provide 3 levels of aggregation.

L0 - Instantaneous readings from hwmon.
L1 - Instantaneous aggregations.  This will be an object that tells you the mix/max/average for the last <period of time>
L2 - Historical aggregations.  This will create objects from the L1 aggregations of N readings every <period of time>. Each object would provide the measure( a datapoint tuple composed of timestamp and value).

The aggregation methods we would be supporting are min,max and average.(For example, the min aggregation method will aggregate the values of different measures to the minimum value of all the measures in time range.)

An openbmc app leveraging dbus-monitor app would be implemented to cater to the above requirements. A yaml based approach will be utilized to define the policy of the app.Inputs to the yaml include:

a) Sensor paths identifying the resource( /sensors/fan_tach/*, sensors/temperature/* ..etc)
b) Granularity, the time between two measures
c) Aggregation Period(the period between which the aggregation method would be invoked)

We would leverage the dbus signals emitted by the hwmon, when the sensor value changes. The application would buffer the measures<timestamp/values> emitted by the signal and invoke the aggregation method based on the period specified in the yaml and the buffer is cleared.

The user would prefer to collect all the aggregates on a hourly/daily basis, so a suggestion for the object path  would be: /xyz/openbmc_project/sensors/aggregation/<time-period>/<type>/<sensor> (example: /xyz/openbmc_project/sensors/aggregation/per-hour/fan_tach/fan1_0)

dkodihal, 2017-09-15:
@tomjoseph83 we may need more updates here on
- the algorithm to calculate averages.
- Do you aggregate per sensor (rg per core temp sensor) or across all core temp sensors, or do both?
- How are the d-bus obects represented. Will min/max/avg be objects or properties on a single object? Which interfaces will the aggregate object implement? How can it reuse the sensor value interface?
-

tomjoseph83, 2017-09-25:
- The average would be calculated proactively whenever the Sensor object emits the signal when the value changes.
-  In the yaml specification, we are giving the provision of aggregating across all core temp sensors and per temp sensor. The current plan is to implement the per sensor.
- The aggregation dbus objects would implement the minimum, maximum and the average aggregation method. The property of each aggregation method is <timestamp, value>

tomjoseph83, 2017-09-25:
@geissonator We can close the design story since we have opened the epic to track the development. #2312

closed 2017-09-27 by rfrandse