openbmc/openbmc #1959: Increase VDDR voltage setpoint for Witherspoon

cdeadmin, 2017-07-13:


geissonator, 2017-07-13:
- [ ] Rename avs workaround script to power workarounds
- [ ] Update script with desired commands from power team (memory volt increase)
- [ ] Update script with desired commands from power team (vdn volt decrease)

rfrandse, 2017-07-28:
<https://gerrit.openbmc-project.xyz/5617> Decrease VDN to .8v for power on
Resolves  #1959 Increase VDDR voltage setpoint for Witherspoon

closed 2017-08-03 by williamspatrick

rkgebhardt, 2017-08-16:
The VDDR rail on Witherspoon is still running at the old level (1.21v) on build 1734B:  ibm-v1.99.8-98-g06734f31-r3
Jordan Keuseman has given me the i2cset commands to use as a workaround.  These do move the VDDR set point to 1.23v, but it does not change the nominal voltages to match those.  This shows up in the  Converged Firmware Toolkit command: mfgcommand -m vapi with the -r (read) and -s (set) commands.  To be able to set the voltage by percentage the nominal voltage for VDDR must be 1.23v.   Is there anything holding this up?