openbmc/openbmc #1960: phosphor-logging: Duplicated Event string in error message

anoo1, 2017-07-13:
The open power error logs have a duplicated Event string in the message that needs to be removed.
This may just need to have the yaml file renamed or moved.

```
   "Id": 6,
    "Message": "org.open_power.Error.Host.Event.Event",
    "Resolved": 0,
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Informational",
    "Timestamp": 1499911463901,
    "associations": []
```

devenrao, 2017-10-23:
**Message: org.open_power.Error.Host.Event.Event**

To remove dumplicate Event the existing error yaml **org/open_power/Host/Event.errors.yaml** need to be renamed as **org/open_power/Host.errors.yaml**

But the problem with the above approach is we already have an error yaml file with the same name **org/open_power/Host.errors.yaml** in openpower-debug-collector repo

As all the error yaml files are copied to the shared location for phosphor-logging to generate elog-errors.hpp error yaml file will be overwritten.

Need to discuss in design meeting

rfrandse, 2017-10-27:
<https://gerrit.openbmc-project.xyz/7498> Rework for change in error yaml for host event error
Resolves  #1960 phosphor-logging: Duplicated Event string in error message

closed 2017-11-03 by geissonator