openbmc/openbmc #1961: [BarrelEye v1.99.8-5] Reset button not handled properly

yadron, 2017-07-13:
If I press "reset", it is not handled properly (the output of journald exactly after reset attempt):

```
Jul 13 15:03:17 barreleye chassis_control.py[876]: Soft Rebooting
Jul 13 15:03:17 barreleye chassis_control.py[876]: Soft off power
Jul 13 15:03:17 barreleye chassis_control.py[876]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Jul 13 15:03:17 barreleye chassis_control.py[876]: Traceback (most recent call last):
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Jul 13 15:03:17 barreleye chassis_control.py[876]:     self._handler(*args, **kwargs)
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/sbin/chassis_control.py", line 186, in softreset_button_signal_handler
Jul 13 15:03:17 barreleye chassis_control.py[876]:     self.softReboot()
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/sbin/chassis_control.py", line 145, in softReboot
Jul 13 15:03:17 barreleye chassis_control.py[876]:     self.softPowerOff()
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/sbin/chassis_control.py", line 121, in softPowerOff
Jul 13 15:03:17 barreleye chassis_control.py[876]:     intf = self.getInterface('host_services')
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/sbin/chassis_control.py", line 94, in getInterface
Jul 13 15:03:17 barreleye chassis_control.py[876]:     obj = bus.get_object(o['bus_name'], o['object_name'], introspect=False)
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
Jul 13 15:03:17 barreleye chassis_control.py[876]:     follow_name_owner_changes=follow_name_owner_changes)
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
Jul 13 15:03:17 barreleye chassis_control.py[876]:     self._named_service = conn.activate_name_owner(bus_name)
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
Jul 13 15:03:17 barreleye chassis_control.py[876]:     self.start_service_by_name(bus_name)
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
Jul 13 15:03:17 barreleye chassis_control.py[876]:     'su', (bus_name, flags)))
Jul 13 15:03:17 barreleye chassis_control.py[876]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Jul 13 15:03:17 barreleye chassis_control.py[876]:     message, timeout)
Jul 13 15:03:17 barreleye chassis_control.py[876]: DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.HostServices was not provided by any .service files
```

Could you please support reset? As workaround, at least it should do "poweroff / poweron" (or even "chassisoff / chassison") one after other.

williamspatrick, 2017-07-17:
@geissonator - Did we miss updating some applications?  Is there already an issue for this?

gtmills, 2018-07-18:
No updates in over a year, Barreleye is no longer a supported machine, and this is a very old build. @geissonator Can we close or is this tracking something?

geissonator, 2018-08-20:
See above.

closed 2018-08-20 by geissonator