openbmc/openbmc #1964: Question about ipmi_get_chassis_cap API

yongli3, 2017-07-14:
There is a TODO  the below sour code, I would like to know what is MRW?
https://github.com/openbmc/phosphor-host-ipmid/blob/master/chassishandler.cpp#L544

bjwyman, 2017-07-14:
MRW = Machine Readable Workbook
https://github.com/open-power/common-mrw-xml
https://github.com/open-power/witherspoon-xml

mdmillerii, 2017-07-14:
The machine readable workbook (MRW) is an xml file that exposes system specific data, including i2c device connections, gpio connections, and ipmi sensor numbers,  to various firmware components, including hostboot, the analogue of bios for OpenPower systems.  The workbook is generally created and edited using the serverwiz gui.https://github.com/open-power/serverwiz

yongli3, 2017-07-17:
Got it! Thanks!

closed 2017-07-17 by yongli3