openbmc/openbmc #1966: External Event Notification

geissonator, 2017-07-14:
Track all work required to allow external applications to register for event notification and then be notified of those events.  

Some notes:
- Plan is some mechanism to subscribe to dbus events to send up to REST web socket
  - Something similar was done on Barreleye (but never merged)
  - Implementation was single direction (not bi-directional)
- external app would connect, and say the things it wants to know about
  - bmc would then send requested alerts over the socket
- Redfish has some alert-like function coming

geissonator, 2017-07-21:
Moving to 5.1, critical for some

bradbishop, 2017-08-03:
Some example code: 

https://bottlepy.org/docs/dev/async.html#finally-websockets

closed 2017-11-13 by zahrens