openbmc/openbmc #1967: BMC switching to alternate bmc chip on reset [Witherspoon]

gkeishin, 2017-07-14:
`Booted with new Master build`

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-8"
VERSION_ID="v1.99.8-8-g87342b9"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-8"
BUILD_ID="v1.99.8"
root@witherspoon:~#
root@witherspoon:~# uptime
 17:34:51 up 5 min,  load average: 0.40, 1.09, 0.64
root@witherspoon:~#
```

`Set the boot side to permanent and reset BMC`

```
Apr 20 18:14:17 witherspoon timemanager[1084]:  User Settings have changed..
Apr 20 18:14:17 witherspoon phosphor-gevent[1109]: ::8039:6a76:8e91:f402%1995782940 - - [2017-04-20 18:14:17] "PUT /org/openbmc/settings/host0/attr/boot_policy HTTP/1.1" 200 169 0.243469
Apr 20 18:14:17 witherspoon phosphor-gevent[1109]: ::8039:6a76:8e91:f402%1995782940 - - [2017-04-20 18:14:17] "POST /login HTTP/1.1" 200 320 0.043649
Apr 20 18:14:17 witherspoon timemanager[1084]:  User Settings have changed..
Apr 20 18:14:17 witherspoon phosphor-gevent[1109]: ::8039:6a76:8e91:f402%1995782940 - - [2017-04-20 18:14:17] "PUT /org/openbmc/settings/host0/attr/boot_flags HTTP/1.1" 200 169 0.237836
Apr 20 18:14:17 witherspoon phosphor-gevent[1109]: ::8039:6a76:8e91:f402%1995782940 - - [2017-04-20 18:14:17] "POST /login HTTP/1.1" 200 320 0.033262
Apr 20 18:14:18 witherspoon phosphor-gevent[1109]: ::8039:6a76:8e91:f402%1995782940 - - [2017-04-20 18:14:18] "POST /org/openbmc/control/bmc0/action/warmReset HTTP/1.1" 200 169 0.284195

Apr 20 18:14:22 witherspoon netipmid[2107]: IPMI GET_SYS_BOOT_OPTIONS
Apr 20 18:14:22 witherspoon netipmid[2107]: IPMID boot option property get: {Default}.
Apr 20 18:14:22 witherspoon netipmid[2107]: Failed to find Sensor Type Default
Apr 20 18:14:22 witherspoon netipmid[2107]: IPMID boot option property get: {PERMANENT}.
Apr 20 18:14:22 witherspoon netipmid[2107]: BootPolicy is [PERMANENT]
Apr 20 18:14:22 witherspoon netipmid[2107]: >> closeSession
Apr 20 18:14:22 witherspoon netipmid[2107]: << closeSession
Apr 20 18:14:23 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
[  OK  ] Stopped Scan FSI devices.
[  OK  ] Removed slice system-fsi\x2dscan.slice.
[  OK  ] Removed slice system-vcs_workaround.slice.
Apr 20 18:14:22 witherspoon systemd[1]: Stopping Network IPMI daemon...
Apr 20 18:14:22 witherspoon dropbear[2189]: Exit (root): Terminated by signal
Apr 20 18:14:22 witherspoon systemd[1]: Removed slice system-op\x2dpower\x2dstop.slice.
Apr 20 18:14:22 witherspoon dropbear[2185]: Exit (root): Terminated by signal
Apr 20 18:14:22 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Apr 20 18:14:22 witherspoon dropbear[2198]: Exit (root): Terminated by signal
Apr 20 18:14:22 witherspoon systemd[1]: Stopped target Start Host0.
Apr 20 18:14:22 witherspoon kernel[943]: [ 2713.370000] watchdog: watchdog0: watchdog did not stop!

```

`boots to alternate bmc chip?`

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-108"
VERSION_ID="v1.99.2-108-g296b27c-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-108"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170223171607
root@witherspoon:~# uptime
 06:17:31 up 12 min,  load average: 0.07, 0.26, 0.32
root@witherspoon:~#
```

gkeishin, 2017-07-14:
- [x]  Serial console 
[Permanent_Serial_Console_log.txt](https://github.com/openbmc/openbmc/files/1149088/Permanent_Serial_Console_log.txt)

gkeishin, 2017-07-16:
Another use case switching the boot

1. Poweron Host
2. Reset BMC

It comes up with the alt side boot
`Reset BMC`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-8"
VERSION_ID="v1.99.8-8-g87342b9"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-8"
BUILD_ID="v1.99.8"
root@witherspoon:~# uptime
 17:33:16 up 4 min,  load average: 1.38, 1.18, 0.55
root@witherspoon:~# [  OK  ] Stopped Wait for /xyz/openbmc_project/state/host0.
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor sysfs LED controller...
...
...
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.10.17-b2cd8101eec3c200940405f12582473617a520f7 (jenkins@openbmc-builder1) (gcc version 6.2.0 (GCC) ) #1 Thu Jul 13 20:20:43 UTC 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d

```
`As the booting kernel continues, kernel takes another reset - refer the log attached and come up with a different kernel i.e on alt side`
```
[    5.220000] lm75 9-004a: hwmon0: sensor 'tmp275'
[    5.540000] occ-hwmon occ1-dev0: occ bus op failed rc:-14
[    5.540000] occ-hwmon occ1-dev0: failed to get OCC poll response: -14
[    5.540000] occ-hwmon: probe of occ1-dev0 failed with error -14

U-Boot 2016.07 (Feb 22 2017 - 17:20:20 +0000)

DRAM:  496 MiB
Flash: 32 MiB
...
....
   Loading Device Tree to 9e9ff000, end 9ea07695 ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 4.7.10-59b0211af5f0057016606aa9cbfb3640ffaba6d4 (jenkins@openbmc-builder1) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 23 17:21:22 UTC 2017
CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d

```
- [x] Serial console Log 
[Poweron_reboot_alt_switch.txt](https://github.com/openbmc/openbmc/files/1151176/Poweron_reboot_alt_switch.txt)

gkeishin, 2017-07-17:
Had a quick talk about this with @amboar and we suspect that the kernel is busted somehow which makes the watchdog kick in and on the AST2500s there's a watchdog feature that boots from the other flash chip if that condition occurs. whilst it's not great that the kernel doesn't finish initializing, 
@shenki @eajames  anything you can dig here.

mdmillerii, 2017-07-17:
Currently neither u-boot or linux are aware of the switch, so on the 2500 the chip selects will be swapped.  On the 2400 one chip will be exposed as two independent devices and chaos will ensue.

What is the state of the watchdog driver and dts?  If the watchdog devices are renamed via the alias nodes but the driver does not parse the attributes and clear alternate boot failover then I would expect alt boot attempt on normal reboot.

gkeishin, 2017-07-17:
@eddiejames  and I were able to recreate this one on the system again.  He is looking into it and updates on this one. Thanks

eddiejames, 2017-07-17:
I was getting all sorts of random warnings (ncsi, occ, etc) on the system George was using.

Boot appears to hang when occ-hwmon fails to probe. If the OCC isn't active, occ-hwmon cannot probe successfully, so that's expected behavior.

Is that expected for a driver probe failure to hang the boot?? Surely not, but I don't see any other output after that.

gkeishin, 2017-07-19:
we hit another another one on hostboot team DD2.0 system today.
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-108"
VERSION_ID="v1.99.2-108-g296b27c-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-108"
BUILD_ID="v1.99.2
```

```
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~#
root@witherspoon:~# cat /etc/version
20170223171607
root@witherspoon:~#
```

geissonator, 2017-07-19:
@eddiejames - Lets specifically track the swapping of bmc flash side (watchdog hit) when we do a bmc reset with the host up and running.

eddiejames, 2017-07-19:
Found the cause of the boot hang, not sure if that is causing the watchdog and side swap though. The SBEFIFO driver hangs if there's no space available in the FIFO, which happens on the second proc SBE if the system isn't booted all the way when the bmc reboots.

Not sure what a good solution here is. Experimenting with timeouts in the SBEFIFO driver.

gkeishin, 2017-07-20:
great @eddiejames  keep posted.

gkeishin, 2017-07-20:
We just hit again another system today...

gkeishin, 2017-07-22:
We hit this in 3 systems again today with the master.

geissonator, 2017-07-24:
@gkeishin - Do we need a separate issue to track the problem with setting the boot policy and that causing a side switch?  Eddie is specifically tracking the issue with doing a r/r with the host running with this issue.

gkeishin, 2017-07-24:
@geissonator Yeah make sense. I ll spun off new one just for RR at host running.

gkeishin, 2017-07-24:
Create one to track RR when host running https://github.com/openbmc/openbmc/issues/2015

eddiejames, 2017-07-24:
#** Status
07/24 - Plan for turning sbefifo into a Linux bus type didn't work at all... Devices still probe automatically during bmc boot while host is on. For some reason FSI automatically scans during boot, it's not done by the FSI scan service.

dkodihal, 2017-07-25:
@geissonator I have a story to move away the boot policy from the old settings interface, but I guess that's now what @gkeishin was looking for?

geissonator, 2017-08-15:
Moving this was to @anoo1 since the boot side selection function is coming from code update team.  @eddiejames fixed the r/r issue in #2015

anoo1, 2017-08-24:
Didn't the issue with today's master and setting the boot option to PERMANENT. Could we re-enable the robot tests to see if we see this again?

gkeishin, 2017-08-25:
I ll run the test and update on it..

geissonator, 2017-09-06:
Guess we're good here @gkeishin ?

gkeishin, 2017-09-07:
Yeah its good now. If we see this again.. I can open one ahead,,

closed 2017-09-07 by gkeishin