openbmc/openbmc #1976: Fan controller crashed on boot [ Witherspoon ]

gkeishin, 2017-07-18:
`Driver info`
```
root@witherspoon:~# uptime
 17:31:29 up 2 min,  load average: 3.59, 1.82, 0.69
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-23"
VERSION_ID="v1.99.8-23-g2ad7ce3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-23"
BUILD_ID="v1.99.8"
root@witherspoon:~# cat /etc/timestamp
20170718013103
root@witherspoon:~#
```

`Journald snippet`

```
Apr 20 17:36:49 witherspoon systemd[1]: Reached target Fan Control Ready.
Apr 20 17:36:49 witherspoon systemd[1]: Started Fan Watchdog Failure Monitor.
Apr 20 17:36:50 witherspoon phosphor-fan-control[2005]: Error in mapper response for getting service name
Apr 20 17:36:50 witherspoon phosphor-fan-control[2005]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Apr 20 17:36:50 witherspoon phosphor-fan-control[2005]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Apr 20 17:36:50 witherspoon phosphor-fan-control[2005]: The operation failed internally.
Apr 20 17:36:50 witherspoon systemd[1]: phosphor-fan-control@0.service: Main process exited, code=dumped, status=6/ABRT
Apr 20 17:36:50 witherspoon systemd[1]: phosphor-fan-control@0.service: Unit entered failed state.
Apr 20 17:36:50 witherspoon systemd[1]: phosphor-fan-control@0.service: Triggering OnFailure= dependencies.
Apr 20 17:36:50 witherspoon systemd[1]: phosphor-fan-control@0.service: Failed with result 'core-dump'.
Apr 20 17:36:50 witherspoon systemd[1]: Starting Fan Watchdog Failure Monitor...
Apr 20 17:36:50 witherspoon systemd[1]: phosphor-fan-control@0.service: Service hold-off time over, scheduling restart

● phosphor-fan-control@0.service - Phosphor Fan Control Daemon
   Loaded: loaded (/lib/systemd/system/obmc-fan-control-ready@0.target.requires/../phosphor-fan-control@.service; disabled; vendor preset: enabled)
  Drop-In: /lib/systemd/system/phosphor-fan-control@0.service.d
           └─fan-watchdog-monitor.conf
   Active: failed (Result: core-dump) since Thu 2017-04-20 17:36:51 UTC; 1min 12s ago
  Process: 2012 ExecStart=/usr/sbin/phosphor-fan-control --control (code=dumped, signal=ABRT)
 Main PID: 2012 (code=dumped, signal=ABRT)

Apr 20 17:36:50 witherspoon systemd[1]: phosphor-fan-control@0.service: Unit entered failed state.
Apr 20 17:36:50 witherspoon systemd[1]: phosphor-fan-control@0.service: Triggering OnFailure= dependencies.
Apr 20 17:36:50 witherspoon systemd[1]: phosphor-fan-control@0.service: Failed with result 'core-dump'.
Apr 20 17:36:51 witherspoon systemd[1]: phosphor-fan-control@0.service: Service hold-off time over, scheduling restart.
Apr 20 17:36:51 witherspoon systemd[1]: Stopped Phosphor Fan Control Daemon.
Apr 20 17:36:51 witherspoon systemd[1]: phosphor-fan-control@0.service: Start request repeated too quickly.
Apr 20 17:36:51 witherspoon systemd[1]: Failed to start Phosphor Fan Control Daemon.
Apr 20 17:36:51 witherspoon systemd[1]: phosphor-fan-control@0.service: Unit entered failed state.
Apr 20 17:36:51 witherspoon systemd[1]: phosphor-fan-control@0.service: Triggering OnFailure= dependencies.
Apr 20 17:36:51 witherspoon systemd[1]: phosphor-fan-control@0.service: Failed with result 'core-dump'.
Warning: phosphor-fan-control@0.service changed on disk. Run 'systemctl daemon-reload' to reload units.
```

`Journald /dmesg logs`
- [x] journald 
[journald_v1.99.8-23-g2ad7ce3_fan_controller_crashed.txt](https://github.com/openbmc/openbmc/files/1154681/journald_v1.99.8-23-g2ad7ce3_fan_controller_crashed.txt)
- [x] dmesg 
[dmesg_v1.99.8-23-g2ad7ce3_fan_controller_crashed.txt](https://github.com/openbmc/openbmc/files/1154684/dmesg_v1.99.8-23-g2ad7ce3_fan_controller_crashed.txt)

gkeishin, 2017-07-18:
@bradbishop ^^

gkeishin, 2017-07-18:
Probably this is because of the above I assume in inventory 
```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan0": {
        "Functional": 1
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan1": {
        "Functional": 1
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan2": {
        "Functional": 1
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/fan3": {
        "Functional": 1
    }
```

spinler, 2017-07-18:
```
    MESSAGE=Error in mapper call to get service name
    PATH=/xyz/openbmc_project/sensors/temperature/ambient
    INTERFACE=xyz.openbmc_project.Sensor.Value
```

spinler, 2017-07-18:
```
root@witherspoon:/sys/class/hwmon# dmesg | grep -i bmp
[    4.780000] bmp280: probe of 3-0077 failed with error -5
```

spinler, 2017-07-18:
I can see the phosphor-hwmon instance getting launched for the sensor, but it keeps dying I think because it couldn't find the sysfs path:

```Jul 17 22:12:18 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40100-dps310\x4076.service: Main process exited, code=exited, status=255/n/a
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000] Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
Jul 17 22:12:18 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40100-dps310\x4076.service: Unit entered failed state.
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
Jul 17 22:12:18 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40100-dps310\x4076.service: Failed with result 'exit-code'.
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Jul 17 22:12:18 witherspoon systemd[1]: Started OpenPower Software Update Manager.
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Jul 17 22:12:18 witherspoon phosphor-network-manager[936]: Error in mapper call
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000] Memory: 430444K/442368K available (4062K kernel code, 280K rwdata, 1192K rodata, 280K init, 111K bss, 11924K reserved, 0K cma-reserved)
Jul 17 22:12:18 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40100-dps310\x4076.service: Service hold-off time over, scheduling restart.
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000] Virtual kernel memory layout:
Jul 17 22:12:18 witherspoon systemd[1]: Started Phosphor Download Manager.
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Jul 17 22:12:18 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Jul 17 22:12:18 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000]     vmalloc : 0x9f800000 - 0xff800000   (1536 MB)
Jul 17 22:12:18 witherspoon systemd[1]: Starting Phosphor Version Software Manager...
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000]     lowmem  : 0x80000000 - 0x9f000000   ( 496 MB)
Jul 17 22:12:19 witherspoon systemd[1]: Started Phosphor Log Manager.
Jul 17 22:12:16 witherspoon kernel[954]: [    0.000000]       .text : 0x80008000 - 0x803ffd20   (4064 kB)
Jul 17 22:12:20 witherspoon phosphor-hwmon-readd[968]: Usage: /usr/sbin/phosphor-hwmon-readd [options]
Jul 17 22:12:20 witherspoon phosphor-hwmon-readd[968]: Options:
Jul 17 22:12:20 witherspoon phosphor-hwmon-readd[968]:     --help               print this menu
Jul 17 22:12:20 witherspoon phosphor-hwmon-readd[968]:     --path=<path>        sysfs location to monitor
Jul 17 22:12:20 witherspoon phosphor-hwmon-readd[968]:     --of-name=<path>     open firmware device name to monitor
Jul 17 22:12:20 witherspoon phosphor-hwmon-readd[968]: ERROR: Path not specified or invalid.
```

spinler, 2017-07-18:
The snippet in the device tree that maps iio drivers to hwmon doesn't list the dps310:
```    iio-hwmon {
        compatible = "iio-hwmon";
        io-channels = <&bmp 1>;
    };```

will investigate how to fix.

gkeishin, 2017-07-18:
Great thanks @spinler

spinler, 2017-07-18:
I have a fix to test, just waiting for a pass 2 machine to become available.

spinler, 2017-07-18:
Tried the following, with no luck
```
	iio-hwmon {
		compatible = "iio-hwmon";
		io-channels = <&bmp 1>, <&dps 1>;
	};

	dps: dps310@76 {
		compatible = "infineon,dps310";
		reg = <0x76>;
		#io-channel-cells = <1>;
	};
```
Also tried 0s for the dps channels, and that also didn't work - nothing showed up in hwmon.

@shenki Can you think of any reason your dsp310 driver wouldn't behave with iio-hwmon?

spinler, 2017-07-19:
Status:
* 7/19: tried several incantations of registering the dps driver with iio-hwmon.  nothing worked

spinler, 2017-07-19:
got the device to show up in hwmon, but now phosphor-hwmon-readd always gets errno 11 (EAGAIN) when trying to read it.  Can read it fine manually

spinler, 2017-07-27:
Will use this issue to track the device tree update so device shows up under hwmon/.  I opened #2038 to track handling EAGAINs in hwmon.  

Device tree update picked up with https://gerrit.openbmc-project.xyz/#/c/5417/

closed 2017-07-27 by spinler