openbmc/openbmc #1978: REST call failure due to "SSL connect error"

samahaba, 2017-07-18:
Driver Info:
=========
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-23"
VERSION_ID="v1.99.8-23-g2ad7ce3"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-23"
BUILD_ID="v1.99.8"
root@witherspoon:#

root@witherspoon:# cat /etc/timestamp
20170718013103

Problem Description
------------------------
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://<SystemName>/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl: (35) SSL connect error

root@witherspoon:# ls -al /proc/*/fd/ | wc -l
ls: /proc/14334/fd/3: cannot read link: No such file or directory
ls: /proc/self/fd/3: cannot read link: No such file or directory
ls: /proc/thread-self/fd/3: cannot read link: No such file or directory
2497
root@witherspoon:#

gkeishin, 2017-07-18:
This is same as https://github.com/openbmc/openbmc/issues/1389  we can close this one

closed 2017-07-18 by gkeishin

sivassrr, 2017-07-18:
This happened immediately after code update, CI also failed.

gkeishin, 2017-07-18:
So we will need the FFDC for this one.. and re-open then.. including the proc fd list output as well.