openbmc/openbmc #1982: phosphor-fan-control crash

shenki, 2017-07-19:
v1.99.8-41-g86a4abc on witherspoon:

```
Jul 19 08:36:55 witherspoon phosphor-fan-control[1331]: Error in mapper call to get service name
Jul 19 08:36:55 witherspoon phosphor-fan-control[1331]: The operation failed internally.
Jul 19 08:36:55 witherspoon phosphor-fan-control[1331]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Jul 19 08:36:55 witherspoon phosphor-fan-control[1331]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Jul 19 08:36:55 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Main process exited, code=dumped, status=6/ABRT
Jul 19 08:36:55 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Unit entered failed state.
Jul 19 08:36:55 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Failed with result 'core-dump'.
Jul 19 08:36:55 witherspoon systemd[1]: phosphor-fan-control-init@witherspoon-fan-watchdog.service: Service hold-off time over, scheduling restart.
```

spinler, 2017-07-19:
Duplicate of #1976 , which is that nobody ever put the dps310 into the iio-hwmon bridge

shenki, 2017-07-19:
Are we going to fix the crash? Perhaps leave this one open so that can be addressed.

shenki, 2017-07-25:
Re-opening to get visibility on the core dump.

spinler, 2017-07-25:
Is this for the same fail and even on the same system as #2020?  We would just need 1 issue.

closed 2017-07-25 by williamspatrick