openbmc/openbmc #1983: Witherspoon fails to boot host

shenki, 2017-07-19:
`obmcutil poweron` from standby, with v1.99.8-41-g86a4abc:

````
ul 19 08:38:06 witherspoon phosphor-host-state-manager[1056]: Host State transaction request
Jul 19 08:38:06 witherspoon systemd[1]: Starting Disable UCD VCS rails...
Jul 19 08:38:06 witherspoon systemd[1]: Starting Wait for Power0 to turn on...
Jul 19 08:38:06 witherspoon systemd[1]: Starting mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition...
Jul 19 08:38:06 witherspoon systemd[1]: Stopped target Chassis0 (Power Off).
Jul 19 08:38:06 witherspoon systemd[1]: Stopped Power off mode bind ir35221 device driver devs.
Jul 19 08:38:06 witherspoon systemd[1]: Stopped target Power0 Off.
Jul 19 08:38:06 witherspoon systemd[1]: Stopped Disable VCS on a power off.
Jul 19 08:38:06 witherspoon systemd[1]: Stopped Power off mode unbind ir35221 device driver devs.
Jul 19 08:38:06 witherspoon systemd[1]: Stopped Stop Power0.
Jul 19 08:38:06 witherspoon systemd[1]: Stopped Reload mboxd after power off.
Jul 19 08:38:06 witherspoon systemd[1]: Stopped Wait for Power0 to turn off.
Jul 19 08:38:06 witherspoon systemd[1]: Stopped target Power0 Off (Pre).
Jul 19 08:38:06 witherspoon systemd[1]: Stopped target Host0 (Stopped).
Jul 19 08:38:06 witherspoon systemd[1]: Stopped target Host0 (Stopping).
Jul 19 08:38:06 witherspoon systemd[1]: Stopped target Stop Host0 (Pre).
Jul 19 08:38:06 witherspoon systemd[1]: Started Phosphor poweron watchdog.
Jul 19 08:38:06 witherspoon systemd[1]: Starting Witherspoon Fan Watchdog Reset...
Jul 19 08:38:06 witherspoon systemd[1]: Starting power on mode unbind ir35221 device driver devs...
Jul 19 08:38:06 witherspoon systemd[1]: Started power on mode unbind ir35221 device driver devs.
Jul 19 08:38:06 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 19 08:38:06 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jul 19 08:38:07 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 19 08:38:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 19 08:38:07 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jul 19 08:38:07 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 19 08:38:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 19 08:38:07 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jul 19 08:38:07 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 19 08:38:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 19 08:38:07 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 19 08:38:07 witherspoon systemd[1]: Started Witherspoon Fan Watchdog Reset.
Jul 19 08:38:07 witherspoon phosphor-chassis-state-manager[1057]: Received signal that power OFF is complete
Jul 19 08:38:07 witherspoon phosphor-chassis-state-manager[1057]: Change to Chassis Power State
Jul 19 08:38:08 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 19 08:38:08 witherspoon kernel[961]: [  340.110000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 19 08:38:08 witherspoon systemd[1]: Started Disable UCD VCS rails.
Jul 19 08:38:08 witherspoon systemd[1]: Starting Perform AVS bus workaround on VRMs...
Jul 19 08:38:08 witherspoon systemd[1]: Started Perform AVS bus workaround on VRMs.
Jul 19 08:38:08 witherspoon systemd[1]: Starting Disable the AVS bus on the VRMs...
Jul 19 08:38:09 witherspoon systemd[1]: Started Disable the AVS bus on the VRMs.
Jul 19 08:38:09 witherspoon systemd[1]: Starting Apply voltage overrides to VRMs...
Jul 19 08:38:09 witherspoon vrm-control.sh[1450]: rail        set        read        current
Jul 19 08:38:09 witherspoon vrm-control.sh[1450]: -------        ------- ------- -------
Jul 19 08:38:10 witherspoon vrm-control.sh[1450]: vdna        0.898V        0.004V        0.000A
Jul 19 08:38:10 witherspoon vrm-control.sh[1450]: vdnb        0.898V        0.004V        0.000A
Jul 19 08:38:10 witherspoon systemd[1]: Started Apply voltage overrides to VRMs.
Jul 19 08:38:10 witherspoon systemd[1]: Starting Enable the AVS bus on VRMs...
Jul 19 08:38:11 witherspoon systemd[1]: Started Enable the AVS bus on VRMs.
Jul 19 08:38:11 witherspoon systemd[1]: Starting Disable VCS before Witherspoon power on...
Jul 19 08:38:11 witherspoon systemd[1]: Started Disable VCS before Witherspoon power on.
Jul 19 08:38:11 witherspoon systemd[1]: Reached target Power0 On (Pre).
Jul 19 08:38:11 witherspoon systemd[1]: Starting Start Power0...
Jul 19 08:38:13 witherspoon systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Jul 19 08:38:13 witherspoon power_control.exe[1052]: PowerControl: setting power up SOFTWARE_PGOOD to 1
Jul 19 08:38:13 witherspoon power_control.exe[1052]: PowerControl: setting power up BMC_POWER_UP to 1
Jul 19 08:38:13 witherspoon systemd[1]: Started Start Power0.
Jul 19 08:38:13 witherspoon timemanager[1083]:  PGOOD has changed..
Jul 19 08:38:13 witherspoon system_manager.py[1007]: Running System State: HOST_POWERING_ON
Jul 19 08:38:36 witherspoon systemd[1]: op-wait-power-on@0.service: Start operation timed out. Terminating.
Jul 19 08:38:36 witherspoon systemd[1]: op-wait-power-on@0.service: Main process exited, code=killed, status=15/TERM
Jul 19 08:38:36 witherspoon systemd[1]: Failed to start Wait for Power0 to turn on.
Jul 19 08:38:36 witherspoon systemd[1]: Dependency failed for Chassis0 (Power On).
Jul 19 08:38:36 witherspoon systemd[1]: Dependency failed for Start Host0.
Jul 19 08:38:36 witherspoon systemd[1]: obmc-host-start@0.target: Job obmc-host-start@0.target/start failed with result 'dependency'.
Jul 19 08:38:36 witherspoon systemd[1]: obmc-host-start@0.target: Triggering OnFailure= dependencies.
Jul 19 08:38:37 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result 'dependency'.
[... 19 lines cut ...]
Jul 19 08:38:37 witherspoon systemd[1]: Stopped Perform AVS bus workaround on VRMs.
Jul 19 08:38:37 witherspoon systemd[1]: Stopped Disable UCD VCS rails.
Jul 19 08:38:37 witherspoon systemd[1]: Stopped Phosphor poweron watchdog.
Jul 19 08:38:37 witherspoon systemd[1]: Started Power off mode unbind ir35221 device driver devs.
Jul 19 08:38:38 witherspoon systemd[1]: Started Wait for Power0 to turn off.
Jul 19 08:38:38 witherspoon systemd[1]: Starting Reload mboxd after power off...
Jul 19 08:38:38 witherspoon systemd[1]: Reached target Power0 Off.
Jul 19 08:38:38 witherspoon systemd[1]: Starting Disable VCS on a power off...
Jul 19 08:38:40 witherspoon power_control.exe[1052]: PowerControl: setting power up SOFTWARE_PGOOD to 0
Jul 19 08:38:40 witherspoon power_control.exe[1052]: PowerControl: setting power up BMC_POWER_UP to 0
Jul 19 08:38:40 witherspoon timemanager[1083]:  PGOOD has changed..
Jul 19 08:38:40 witherspoon mboxctl[1552]: Reset: Success
Jul 19 08:38:39 witherspoon systemd[1]: Started Reload mboxd after power off.
Jul 19 08:38:39 witherspoon systemd[1]: Started Disable VCS on a power off.
Jul 19 08:38:39 witherspoon systemd[1]: Starting Power off mode bind ir35221 device driver devs...
Jul 19 08:38:39 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jul 19 08:38:39 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jul 19 08:38:39 witherspoon systemd[1]: Started Power off mode bind ir35221 device driver devs.
Jul 19 08:38:39 witherspoon systemd[1]: Reached target Chassis0 (Power Off).
Jul 19 08:38:39 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jul 19 08:38:40 witherspoon phosphor-chassis-state-manager[1057]: Received signal that power OFF is complete
Jul 19 08:38:40 witherspoon phosphor-chassis-state-manager[1057]: Change to Chassis Power State
Jul 19 08:38:40 witherspoon system_manager.py[1007]: Running System State: HOST_POWERING_OFF
Jul 19 08:38:40 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
```

gkeishin, 2017-07-19:
I have the same driver and booted good on FTC system
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-41"
VERSION_ID="v1.99.8-41-g86a4abc"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-41"
BUILD_ID="v1.99.8"
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:~#

Ubuntu Zesty Zapus (development branch) 05425 hvc0

05425 login: root
Password:
Last login: Wed Jul 19 00:36:09 CDT 2017 on hvc0
Welcome to Ubuntu Zesty Zapus (development branch) (GNU/Linux 4.10.0-gd46959c ppc64le)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
root@05425:~#
```

amboar, 2018-04-18:
Closing this as it's obsolete.

closed 2018-04-18 by amboar