openbmc/openbmc #1984: [xCAT]Can not activate a PNOR image if another image already activated.

gurevichmark, 2017-07-19:
Images loose their attributes after activation of an uploaded image.

Steps to recreate:

1. Upload and activate a pnor.tar image (`redbud`). Wait for the image to become active.
2. Upload another pnor.tar image (`sequoia`) with a different version number. The software enumeration should show both images. One active and one ready:
```
[root@stratton01 xcat]# curl -b cjar -k -H "Content-Type: application/json" -X GET https://<BMC IP>/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/2a1022fe": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Ready",
      "ExtendedVersion": "op-build-v1.17-176-g11dd356-dirty,buildroot-2017.05-93-g9b91216,skiboot-v5.7-rc2,hostboot-3fa8be9,linux-4.12.1-openpower1-p3a0d8f7,petitboot-v1.5.0-p3eae874,machine-xml-efdf4aa,occ-1647d53,hostboot-binaries-6a7251e,capp-ucode-9c73e9f,sbe-02021c6-p1db02ba",
      "Path": "/tmp/images/2a1022fe",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.67"
    },
    "/xyz/openbmc_project/software/32f129cd": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.17-177-gdd8f32b-dirty,buildroot-2017.05-93-g9b91216,skiboot-v5.7-rc2-p7582c9b,hostboot-3fa8be9,linux-4.12.1-openpower1-p899da6c,petitboot-v1.5.0-pc8fef8e,machine-xml-efdf4aa,occ-1647d53,hostboot-binaries-6a7251e,capp-ucode-9c73e9f,sbe-02021c6-p056301c",
      "Path": "/tmp/images/32f129cd",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active",
      "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.17_1.68"
    },
    "/xyz/openbmc_project/software/865701de": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.8-18-g7778072-dirty"
    }
  },
  "message": "200 OK",
  "status": "ok"
}[root@stratton01 xcat]#
```

3. Activate the second image. 
```
 curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data": "xyz.openbmc_project.Software.Activation.RequestedActivations.Active"}' https://<BMC IP>/xyz/openbmc_project/software/2a1022fe/attr/RequestedActivation
```

4. The enumeration shows both images lost their attributes:

```
[root@stratton01 xcat]#  curl -b cjar -k -H "Content-Type: application/json" -X GET https://<BMC IP>/xyz/openbmc_project/software/enumerate
{
  "data": {
    "/xyz/openbmc_project/software/2a1022fe": {
      "Path": "/tmp/images/2a1022fe",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.17_1.67"
    },
    "/xyz/openbmc_project/software/32f129cd": {
      "Path": "/tmp/images/32f129cd",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "Version": "IBM-witherspoon-redbud-ibm-OP9_v1.17_1.68"
    },
    "/xyz/openbmc_project/software/865701de": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "Path": "",
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "v1.99.8-18-g7778072-dirty"
    }
  },
  "message": "200 OK",
  "status": "ok"
}[root@stratton01 xcat]#
```

saqibkh, 2017-07-19:
Seems to be related to https://github.com/openbmc/openbmc/issues/1917 which @anoo1 fixed recently with a work around. 
Maybe we need that work around in other places too.

mtritz, 2017-07-24:
I was not able to replicate this issue. With latest image, I was able to activate two images with attributes intact. Are you seeing the issue consistently? If so, could you please provide more details about your testing environment?

mtritz, 2017-07-24:
# **STATUS**
7/24: Issue is related to ActivationProgress - during the final steps of the activation process, setting progress to 90 or 100 on the second image crashes the host software updater and produces a core dump. Will investigate solutions tomorrow.

closed 2017-07-25 by williamspatrick