openbmc/openbmc #1985: Add gpio deglitch support to fan presence

bradbishop, 2017-07-19:
Add a yaml parameter to the gpio sensor type that allows a deglitch to be configured.

amboar, 2017-07-20:
@bradbishop can you expand on what this means? Is this using the hardware debounce? Or trying to do it in software somehow?

mdmillerii, 2017-07-20:
Same thought as @amboar  ... although looking around I realize these fan presence gpios are in the pending patch to add [gpio-keys-polled](http://patchwork.ozlabs.org/patch/791208/).  Is this proposing a pure user space solution for deglitch?

Looking at [the doc](https://github.com/openbmc/linux/blob/dev-4.10/Documentation/devicetree/bindings/input/gpio-keys-polled.txt#L28) `gpio-keys-polled` already supports a `debounce-interval` in the kernel driver and this should just be a device tree property not yaml.

Note: the default is 5ms and is rounded up to a multiple of the poll interval.

amboar, 2017-07-20:
Note I brain-farted the clock association in the GPIO driver, so the debounce calculations will be wrong. I need to send patches to rectify (timings are derived from PCLK, not HPLL).

@mdmillerii: right - the devicetree support for `debounce-interval` made me wonder why this issue was described the way it is. If devicetree is the direction we're going we can't yaml it.

spinler, 2017-07-20:
We're more concerned about deglitching for meatspace time - if a human happens to jiggle a FRU while they're inserting it, we want to be able to handle that.  Can we increase the debounce-interval to be something a lot larger, like 1s (or maybe a bit less)?

mdmillerii, 2017-07-20:
yes, its a simple counter of the consecutive reads per poll for polled keys.  its 32 bit signed int for the counter & threshoold, and the threshold is calculated in ms from u32 (32 bit integers).

bradbishop, 2017-08-24:
Deglitch is just a DTS update (debounce-interval) so no plumbing is needed for general use.

As such, this is a Witherspoon specific story, and no deglitching support (beyond what is done by PDM) is required. Closing...

Also this was a dup of openbmc/openbmc#1861.

closed 2017-08-24 by bradbishop