openbmc/openbmc #1987: Update 9552 LED names in MRW to match device tree

spinler, 2017-07-19:
The driver prepends 'pca955x:' to the names, so the MRW will have to match that so the led code can find them.

bjwyman, 2017-07-20:
See 1738 comment from Milton. Thinks label should be `pca955x:amber:fan0`.

mdmillerii, 2017-07-20:
We also need to fix the directly attached leds to the naming format.

spinler, 2017-07-25:
The latest MRW fails our builds because of some empty sensor ID fields on the GPU targets.  Have requested the owner to fix those, and then I can fix the LEDs.

spinler, 2017-08-10:
For reasons I don't completely understand, the latest drivers don't even have the 'pca955x:' in sysfs anymore:
```
# ls /sys/class/leds/
fan0         fan1         fan2         fan3         front-fault  front-id     front-power  rear-fault   rear-id      rear-power
```

and all of the LEDs show up in D-Bus now.
I guess I will close this.

closed 2017-08-10 by spinler

mdmillerii, 2017-08-10:
That would be because they changed from the leds-pca955x driver to the leds-gpio driver.  It doesn't tack on the name.  It also can not support hardware blink and will rely on a kernel timer to change the led state.