openbmc/openbmc #1988: Investigate adding hardware blink support to the pca955x driver

spinler, 2017-07-19:
The chip supports blinking in the hardware, so it would be nice if the driver could make use of it.

shenki, 2017-07-24:
The driver supports this already. Please see the bindings at Documentation/devicetree/bindings/leds/common.txt and Documentation/devicetree/bindings/leds/leds-pca9532.txt.

mdmillerii, 2017-07-24:
@shenki leds-pca9532 supports it but leds-pca955x does not.  

This requests the support be added to that instance of the driver.

shenki, 2017-07-24:
I see. Looking at the pca955x driver, it does appear to support blinking:

```
static int pca955x_led_set(struct led_classdev *led_cdev,
                            enum led_brightness value)
{

 ...

        switch (value) {
        case LED_FULL:
                ls = pca955x_ledsel(ls, ls_led, PCA955X_LS_LED_ON);
                break;
        case LED_OFF:
                ls = pca955x_ledsel(ls, ls_led, PCA955X_LS_LED_OFF);
                break;
        case LED_HALF:
                ls = pca955x_ledsel(ls, ls_led, PCA955X_LS_BLINK0);
                break;
```

@spinler can you confirm that you tested this and it did not work?

spinler, 2017-07-25:
Haven't tried it.  Just opened this based on @mdmillerii 's comments in #1738.  Will attempt to find some time to try it.

spinler, 2017-07-25:
All 7 front LEDs blink happily away.    @shenki if you're saying this is done under HW control go ahead and close this.

legoater, 2017-07-25:
I suppose we can send the 955x patchset to mainline now. I am not sure the GPIO part was tested ?

mdmillerii, 2017-07-25:
The leds will blink under sw control with the same user-space ABI.  One would have to search the timer list or crash the kernel and test if the lights continue to blink to differentiate hw vs hw blink.

shenki, 2017-09-25:
There is no outstanding kernel work for this

closed 2017-09-25 by shenki

mdmillerii, 2017-09-25:
I'd argue that the work was not done but the issue is mute because they switched to gpio leds which can't support hardware blink.