openbmc/openbmc #1991: System fails to boot - phosphor-fan-monitor sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure

geissonator, 2017-07-19:
Getting this with latest master.  We already have #1976 but we seemed to boot ok with that.  Now we don't seem to boot at all.

```
Apr 20 18:41:51 witherspoon phosphor-fan-monitor[1578]: Error in mapper call to get service name
Apr 20 18:41:51 witherspoon phosphor-fan-monitor[1578]: The operation failed internally.
Apr 20 18:41:51 witherspoon phosphor-fan-monitor[1578]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Apr 20 18:41:51 witherspoon phosphor-fan-monitor[1578]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Apr 20 18:41:51 witherspoon systemd[1]: phosphor-fan-monitor@0.service: Main process exited, code=dumped, status=6/ABRT
Apr 20 18:41:51 witherspoon systemd[1]: Failed to start Phosphor Fan Monitor Daemon.
Apr 20 18:41:51 witherspoon systemd[1]: Dependency failed for Fan Control Ready.
Apr 20 18:41:51 witherspoon systemd[1]: obmc-fan-control-ready@0.target: Job obmc-fan-control-ready@0.target/start failed with result 'dependency'.
Apr 20 18:41:51 witherspoon systemd[1]: phosphor-fan-monitor@0.service: Unit entered failed state.
Apr 20 18:41:51 witherspoon systemd[1]: phosphor-fan-monitor@0.service: Triggering OnFailure= dependencies.
Apr 20 18:41:51 witherspoon systemd[1]: phosphor-fan-monitor@0.service: Failed with result 'core-dump'.
...
Apr 20 18:41:55 witherspoon phosphor-fan-control[1613]: Error in mapper call to get service name
Apr 20 18:41:55 witherspoon phosphor-fan-control[1613]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Apr 20 18:41:55 witherspoon phosphor-fan-control[1613]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
Apr 20 18:41:55 witherspoon phosphor-fan-control[1613]: The operation failed internally.
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Main process exited, code=dumped, status=6/ABRT
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Unit entered failed state.
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Triggering OnFailure= dependencies.
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Failed with result 'core-dump'.
Apr 20 18:41:55 witherspoon systemd[1]: Starting Fan Watchdog Failure Monitor...
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Service hold-off time over, scheduling restart.
Apr 20 18:41:55 witherspoon systemd[1]: Stopped Phosphor Fan Control Daemon.
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Start request repeated too quickly.
Apr 20 18:41:55 witherspoon systemd[1]: Failed to start Phosphor Fan Control Daemon.
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Unit entered failed state.
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Triggering OnFailure= dependencies.
Apr 20 18:41:55 witherspoon systemd[1]: phosphor-fan-control@0.service: Failed with result 'core-dump'.
Apr 20 18:41:56 witherspoon systemd[1]: Started Fan Watchdog Failure Monitor.
Apr 20 18:42:04 witherspoon phosphor-dbus-monitor[1047]: Shutting down system.  There are not enough functional fans.
```

```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-55"
VERSION_ID="v1.99.8-55-gecf5269"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-55"
```

Missing hwmon maybe?
```
[witherspoon]# ps -ef | grep hwmon
  958 root       0:17 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/max31785@52
 1679 root       0:00 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@70
 1684 root       0:00 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@71
 1689 root       0:00 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@71
 1690 root       0:00 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@70
 1693 root       0:00 grep hwmon
```

```
[witherspoon]# tracg | grep hwmon
Apr 20 17:29:33 witherspoon kernel: lm75 9-004a: hwmon0: sensor 'tmp275'
Apr 20 17:29:49 witherspoon kernel[960]: [    3.950000] lm75 9-004a: hwmon0: sensor 'tmp275'
Apr 20 17:29:55 witherspoon phosphor-hwmon-readd[974]: Usage: /usr/sbin/phosphor-hwmon-readd [options]
Apr 20 17:29:55 witherspoon phosphor-hwmon-readd[974]: Options:
Apr 20 17:29:55 witherspoon phosphor-hwmon-readd[974]:     --help               print this menu
Apr 20 17:29:55 witherspoon phosphor-hwmon-readd[974]:     --path=<path>        sysfs location to monitor
Apr 20 17:29:55 witherspoon phosphor-hwmon-readd[974]:     --of-name=<path>     open firmware device name to monitor
Apr 20 17:29:55 witherspoon phosphor-hwmon-readd[974]: ERROR: Path not specified or invalid.
Apr 20 17:30:03 witherspoon phosphor-hwmon-readd[1009]: Usage: /usr/sbin/phosphor-hwmon-readd [options]
```

bradbishop, 2017-07-20:
The root cause here seems to be a downlevel 0x3030 firmware on the max chip on the fan card.

spinler, 2017-07-20:
The system wsbmc004 has the latest fan HW, and an old enough ambient sensor that those don't cause problems.  it dies because presence-detect is never creating the Present property for any fan in the inventory.

rfrandse, 2017-07-21:
<https://gerrit.openbmc-project.xyz/5406> wspoon: Normalize native fan bbappend search paths
Resolves:  #1991 System fails to boot - phosphor-fan-monitor sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure

closed 2017-07-21 by williamspatrick