openbmc/openbmc #1992: "500 Internal Server Error" rettreiving BootStatus

stewart-ibm, 2017-07-20:
When host was running and maxing out console output:

```Running tests...
----------------------------------------------------------------------
OpTestSystem START STATE: 0
curl   -b cjar  -c cjar   -k  -H  'Content-Type: application/json'   -d '{"data" : []}'  -X POST https://spoon2-bmc///org/openbmc/control/chassis0/action/powerOff
OpTestSystem TRANSITIONED TO: 7
curl   -b cjar   -k  -H  'Content-Type: application/json'   -d '{"data" : []}'  -X GET https://spoon2-bmc///org/openbmc/sensors/ho
st/BootProgress
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2275  100  2262    0    13     57      0  0:00:39  0:00:39 --:--:--   462
{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, th
e message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 858, in _handle", 
      "    route, args = self.router.match(environ)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1083, in custom_router_match", 
      "    route.callback._setup(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 171, in _setup", 
      "    return self.setup(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 492, in setup", 
      "    request.route_data['map'] = self.find(path, callback)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 483, in find", 
      "    path=path)}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 204, in try_mapper_call", 
      "    return f(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 60, in get_object", 
      "    retries, interval)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 42, in retry", 
      "    return func()", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 59, in <lambda>", 
      "    path, interfaces, signature='sas'),", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application
 did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was
 broken."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
}
```

On a witherspoon with:
```root@spoon2-bmc:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-8"
VERSION_ID="v1.99.8-8-g87342b9"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-8"
BUILD_ID="v1.99.8"
```

stewart-ibm, 2017-07-20:
After that, machine doesn't power off, even after repeatedly issuing "obmcutil poweroff" and waiting minutes.

stewart-ibm, 2017-07-20:
For reference, I ran `hexdump -C -v /dev/zero` from the petitboot shell

stewart-ibm, 2017-07-20:
```
Jul 20 02:17:05 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:05] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.557797
Jul 20 02:17:11 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:11] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.055207
Jul 20 02:17:16 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:16] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.054804
Jul 20 02:17:17 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:17] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.055523
Jul 20 02:17:22 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:22] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.055122
Jul 20 02:17:27 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:27] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.055672
Jul 20 02:17:33 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:33] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.054432
Jul 20 02:17:36 spoon2-bmc phosphor-host-state-manager[1050]: Host State transaction request
Jul 20 02:17:36 spoon2-bmc systemd[1]: Starting Soft power off of the host...
Jul 20 02:17:36 spoon2-bmc ipmid[1044]: Pushing cmd on to queue
Jul 20 02:17:36 spoon2-bmc ipmid[1044]: Asserting SMS Attention
Jul 20 02:17:36 spoon2-bmc ipmid[1044]: SMS Attention asserted
Jul 20 02:17:38 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:38] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.122995
Jul 20 02:17:39 spoon2-bmc phosphor-host-state-manager[1050]: Host State transaction request
Jul 20 02:17:39 spoon2-bmc ipmid[1044]: [[0;1;31mHost control timeout hit![[0m
Jul 20 02:17:39 spoon2-bmc ipmid[1044]: Timer expired
Jul 20 02:17:39 spoon2-bmc phosphor-softpoweroff[3528]: Host control signal values
Jul 20 02:17:39 spoon2-bmc phosphor-softpoweroff[3528]: Timeout on host attention, continue with power down
Jul 20 02:17:40 spoon2-bmc systemd[1]: Started Soft power off of the host.
Jul 20 02:17:43 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:43] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.055267
Jul 20 02:17:49 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:49] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.056072
Jul 20 02:17:54 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:54] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.055160
Jul 20 02:17:59 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:17:59] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.063257
Jul 20 02:18:05 spoon2-bmc phosphor-gevent[3369]: ::80d9:6176:8e91:f402%1995234076 - - [2017-07-20 02:18:05] "GET ///org/openbmc/s
ensors/host/BootProgress HTTP/1.1" 200 246 0.337298
```

stewart-ibm, 2017-07-20:
[spoon2-log.txt](https://github.com/openbmc/openbmc/files/1160964/spoon2-log.txt)

geissonator, 2017-07-20:
The soft power off just keeps repeating, that's weird.

If you've got org/openbmc in a curl command then there's a good chance it's been deprecated (we're removing all and going to xyz/openbmc_project/).  I have a feeling using /org/openbmc/control/chassis0/action/powerOff could def have caused some issues here.

stewart-ibm, 2017-07-20:
I did repeatedly try to power off things...

r.e. deprecated commands, *please* don't break op-test-framework. or at least have a decently long deprecation period so that every bmc can be updated to something that doesn't regress elsewhere...

stewart-ibm, 2017-07-20:
```
root@romulus:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-3"
VERSION_ID="v1.99.7-3-gd0c8db0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-3"
BUILD_ID="v1.99.7"
root@romulus:~# time obmcutil power
pgood = 1
state = 1
pgood_timeout = 10

real    0m20.036s
user    0m4.680s
sys     0m0.350s

root@romulus:~# journalctl -f
-- Logs begin at Mon 2017-07-17 06:38:07 UTC. --
Jul 20 04:23:34 romulus obmc-console-server[837]: obmc-console-server: Write error: Resource temporarily unavailable
Jul 20 04:23:35 romulus obmc-console-server[837]: obmc-console-server: Write error: Resource temporarily unavailable
Jul 20 04:23:36 romulus obmc-console-server[837]: obmc-console-server: Write error: Resource temporarily unavailable

  489     1 root     R    38448   9%  48% /lib/systemd/systemd-journald
 8274     2 root     SW       0   0%  19% [kworker/u2:2]
  837     1 root     S     1852   0%  14% /usr/sbin/obmc-console-server --config /etc/obmc-console.conf ttyVUART0
 8276     2 root     RW       0   0%  12% [kworker/u2:4]
```

Even after issuing poweroff, `obmcutil power` is taking in excess of 17 seconds, but it *does* turn off.

stewart-ibm, 2017-07-20:
Romulus *did* turn off, and I can reproduce the `500 Internal Server Error` there too.

gtmills, 2018-10-12:
@stewart-ibm Still a problem?

closed 2019-05-16 by stale[bot]