openbmc/openbmc #1993: openpower-occ-control: object not created on BMC reboot or service restart

mine260309, 2017-07-20:
In `openpower-occ-control`, the objects `/org/open_power/control/occ?` are created with match of below objects are added:
```
/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu?
```
When the object is not there and host starts, the above object will be created by inventory manager, and `openpower-occ-control` will receive this signal to create objects.

But when the object is already there, e.g. host is already on, the signal will never be received by `openpower-occ-control` and thus the objects will never be created.

This is common, e.g.
1. BMC reboot when host is on.
2. The service is restarted (either manually restarted by user or crashed and restarted by systemd).

And it causes the occ control objects are not created in such cases.

`openpower-occ-control` needs a way to get notified when the object exists, instead of it's added.

geissonator, 2017-07-20:
I think we have a few issues on this already, @dkodihal has a fix coming I believe.

dkodihal, 2017-07-20:
This will be fixed via #1824.

closed 2017-07-20 by dkodihal