openbmc/openbmc #1994: console drops charactes in hexdump of 32k

stewart-ibm, 2017-07-20:
On Witherspoon:
```
root@spoon2-bmc:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-8"
VERSION_ID="v1.99.8-8-g87342b9"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-8"
BUILD_ID="v1.99.8"
```

If a test of outputting a bunch of data to console and checking it all came out is done, we fail. This is now in upstream op-test-framework as of f1cd0090216d550db8b27063d1050c24d7c1e979

```
./op-test ..... --run testcases.Console
```
<SNIP>
```
======================================================================
FAIL [6.790s]: runTest (testcases.Console.Console32k)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/stewart/op-test-framework/testcases/Console.py", line 46, in runTest
    self.assertTrue( len(zeros) == 3+(count*bs)/16, "Unexpected length of zeros %u" % (len(zeros)))
AssertionError: Unexpected length of zeros 1637

----------------------------------------------------------------------
```

stewart-ibm, 2017-07-20:
FYI I've pointed @jk-ozlabs to the test case

pridhiviraj, 2017-07-28:
I have observed similar issue when just doing cat of opal msg log on to the console.
Seeing lot of characters dropped and stops in b/w like below.

Instance 1:
[   45.172289552,7] OPAL: Start CPU 0x0031 (PIR 0x0031) -> 0x000000000000a6c8
[   45.172616528,7] OPAL: Start CPU 0x0032 (PIR 0x0032) -> 0x000000000000a6c8
[   45.172994272,7] OPAL: Start CPU 0x0033 (PIR 0x0033) -> 0x000000000

Instance 2:
[   17.178907392,7] XIVE[ IC 00  ] [reset] disabling source 0x3e
[   17.178910624,7] XIVE[ IC 00  ] [reset] disabling source 0x3f
[   17.185494384,7] XIVE[ IC 08  ] [reset] disabling source 0x100000
[   1

It is not dumping full log to the console.

closed 2019-04-27 by stale[bot]