openbmc/openbmc #1996: How to get the request data length in the ipmid_callback_t callback function?

yongli3, 2017-07-20:
There is a ipmi_request_t par parameter in the ipmid_callback_t function, it is a void *. I find I cannot check the request data length in this callback function. 
Such as the "chassis identify" command. If the client send with the "force" option, the request data will be 2 bytes.

tomjoseph83, 2017-07-31:
@yongli3 This is a shortcoming in the current implementation where the request packet length is not passed to the command handler. We have plans to refactor the provider library implementation, since it is based on the reference implementation done for Barreleye. We can handle this as part of refactoring.

yongli3, 2017-07-31:
@tomjoseph83 Thanks for your input. Could you share more information about the refactor? is there any related PR? or schedule/roadrmap/wiki?

tomjoseph83, 2017-08-01:
@yongli3 https://github.com/openbmc/openbmc/issues/779

Let me know if you need more info.

tomjoseph83, 2017-08-02:
@yongli3 Please close the issue.

closed 2017-08-03 by yongli3