openbmc/openbmc #1997: Mapper Unknown lvalue '--blacklists' in section 'Service' [Witherspoon]

gkeishin, 2017-07-20:
```
Jul 19 13:04:43 witherspoon systemd[1]: [/lib/systemd/system/xyz.openbmc_project.ObjectMapper.service:9] Unknown lvalue '--blacklists' in section 'Service'
```
```
root@witherspoon:~# cat /lib/systemd/system/xyz.openbmc_project.ObjectMapper.service
[Unit]
Description=Phosphor DBus Service Discovery Manager
Before=obmc-mapper.target

[Service]
Restart=always
Type=dbus
ExecStart=/usr/sbin/phosphor-mapper \
        --path_namespaces="${MAPPER_NAMESPACES}" \
        --interface_namespaces="${MAPPER_INTERFACES}"
        --blacklists="${MAPPER_BLACKLISTS}" \
        --interface_blacklists="${MAPPER_INTERFACEBLACKLISTS}"
BusName=xyz.openbmc_project.ObjectMapper
TimeoutStartSec=300
EnvironmentFile=/etc/default/obmc/mapper
Environment="PYTHONUNBUFFERED=1"

[Install]
WantedBy=obmc-standby.target
root@witherspoon:~#

```

gkeishin, 2017-07-20:
https://gerrit.openbmc-project.xyz/#/c/5407/

sivassrr, 2017-07-21:
@gkeishin Are we seeing it in the latest master "Build #899-4403eb0a631434 (Jul 20, 2017 4:14:14 PM) "

rfrandse, 2017-07-21:
<https://gerrit.openbmc-project.xyz/5407> Fix mapper args syntax error
Resolves  #1997 Mapper Unknown lvalue '--blacklists' in section 'Service' [Witherspoon]

closed 2017-07-21 by williamspatrick