openbmc/openbmc #1998: Missing RSA and DSA dropbear key

gkeishin, 2017-07-20:
```
Jul 19 13:04:11 witherspoon systemd[1]: Started SSH Per-Connection Server (xx.xx.xx.xx:42485).
Jul 19 13:04:12 witherspoon dropbear[1087]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jul 19 13:04:12 witherspoon dropbear[1087]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jul 19 13:04:12 witherspoon dropbear[1087]: Child connection from ::ffff:xx.xx.xx.xx:42485

root@witherspoon:~# ls -lart /etc/dropbear/
total 1
drwxr-xr-x    1 root     root             0 Apr 20 17:31 .
-rw-------    1 root     root           805 Jun 29 13:22 dropbear_rsa_host_key
drwxr-xr-x    1 root     root             0 Jul 19 13:04 ..
root@witherspoon:~#
```

I added those keys and now it doesn't drop

```
Jul 19 13:10:48 witherspoon systemd[1]: Started SSH Per-Connection Server (xx.xx.xx.xx:42521).
Jul 19 13:10:49 witherspoon dropbear[1120]: Child connection from ::ffff:xx.xx.xx.xx:42521
Jul 19 13:10:52 witherspoon dropbear[1120]: PAM password auth succeeded for 'root' from ::ffff:xx.xx.xx.xx:42521

root@witherspoon:~# ls -lart /etc/dropbear/
total 2
-rw-------    1 root     root           805 Jun 29 13:22 dropbear_rsa_host_key
drwxr-xr-x    1 root     root             0 Jul 19 13:04 ..
-rw-------    1 root     root           458 Jul 19 13:08 dropbear_dss_host_key
-rw-------    1 root     root           242 Jul 19 13:10 dropbear_ecdsa_host_key
drwxr-xr-x    1 root     root             0 Jul 19 13:10 .
root@witherspoon:~# 
```

gkeishin, 2017-07-20:
We may need to add
-[x] dropbearkey -t ecdsa -f /etc/dropbear/dropbear_ecdsa_host_key
-[x] dropbearkey -t dss -f /etc/dropbear/dropbear_dss_host_key

gkeishin, 2017-07-20:
Currently the code had only to generate RSA keys

```
[Unit]
Description=SSH Key Generation
ConditionPathExists=|!/etc/dropbear/dropbear_rsa_host_key

[Service]
Type=oneshot
ExecStart=@SBINDIR@/dropbearkey -t rsa -f /etc/dropbear/dropbear_rsa_host_key
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```

sivassrr, 2017-07-21:
@gkeishin  Is this related to latest master testing "Build #899-4403eb0a631434 (Jul 20, 2017 4:14:14 PM) "

gkeishin, 2017-07-21:
This is generic observation for all driver so far.

gkeishin, 2017-09-19:
@ratagupt  ^^^

rfrandse, 2018-02-15:
<https://gerrit.openbmc-project.xyz/9005> dropbear: Create rsa host key in user specified path vs /etc/dropbear
Resolves  #1998 Missing DSS and DSA dropbear key

closed 2018-03-01 by geissonator