openbmc/openbmc #1999: Sprint 17.07.26 - Committed

rfrandse, 2017-07-20:


dkodihal, 2017-07-24:
***Squad 1***

**Ratan**
- #957 Cache host inventory in filesystem
Defects
- #1889 REST Post method unable to handle BYTE data

**Jayanth**
- TBD

**Vishwa**
- #1714 REST API to set a user login password (split?)
Defects

**Dhruv**
- #1357 Inventory FRU and Core is set "Functional" but not "Present"
- #1211 boot progress and inventory states for cores
Defects

**Tom**
- #1957 DCMI: Design app for data sampling/telemetry
Defects
- #1548 [xCAT]How to set boot media and boot mode

**Devender**
Vacation 27th Jul - 3rd Aug
Come back and pick from backlog.

**Deepak**
Moved from 07.12
- #1856 DCMI: Design app for data sampling/telemetry
- #2008 Interface and implementation of the boot policy setting
Defects
- #

geissonator, 2017-07-24:
**Squad 2**

**Andrew:geissonator -**

 **17.07.26:**

- #1676 (reboot count decrement)
- #1959 (vddr voltage setpoint increases)
- #1958 (voltage power on workarounds)
- #1709 (xcat inventory issues)

RAS error inject defect investigation
https://gerrit.openbmc-project.xyz/#/c/3749/ (old inventory removal)

-------------------------------------------------------------------------------------------------

**Gunnar:gtmills:**
 **17.06.28:**

  - **#1329 (power supply presence)**

 **17.07.12:**

  - #1860 (PDM elogs)

-------------------------------------------------------------------------------------------------
**Chris:cbostic**: 

**17.06.28:**

- **#1783 (witherspoon gpio-keys)**

 **17.07.12:**

- #1727 (ucd driver support)

------------------------------------------------------------------------------------------------
**Brandon:bjwyman**: 80%

 **17.07.12:**
- #1835 (fan control final piece) -> Back to Barth

 **17.07.26:**
 - #1731 (power supply monitor for faults)

-------------------------------------------------------------------------------------------------
**Eddie:eddiejames**: 80%

**17.06.28:**

- #1883 (hwmon skeleton for power supply driver)

 **17.07.26:**

  - BMC r/r
  - Button support (power on/off and reset)

-------------------------------------------------------------------------------------------------

**Matt S:spinler: 80%** 
**17.06.28:**

- #1724 (chip fault analysis)

- **17.07.26:**
  - #1627 (systemd journal persistence and size restriction)

-------------------------------------------------------------------------------------------------
**Matt B:msbarth: 80%**
**17.06.28:**

- #1835 (delay on events parameter)

- **#1839 (handle scale property)**
  - Lower priority, finish as time allows

- #1792 (yaml updates)

Focus on smart fan speed integration, need to ensure we don't destabilize witherspoon beta

-------------------------------------------------------------------------------------------------
**Brad:bradbishop**: 70%

**17.05.31:**
- **#1318 (gpio presence)**

 **17.07.26:**
  - Continue helping on code update squad

-------------------------------------------------------------------------------------------------
**Dinesh:chinaridinesh**: 65%

**17.06.14:**
- **#1315 (Ensure uboot fw_env update script uses new inventory object)**

geissonator, 2017-07-24:
# Squad 3 -

**Adriana:anoo1 -**
**17.06.14**
- #1551 mboxd: Use tmp location for pnor patches
  - Code up for review - https://gerrit.openbmc-project.xyz/#/c/5432/

**17.07.12**
- **#1940 BMC: Ability for u-boot to execute the kernel that's in a ubi volume**
- **#1943 BMC: Tell u-boot where squashfs image is**
- #1333: PNOR: Enable patching for lab/mfg

**17.07.26**

---

**Brad:bradbishop -**
**17.06.28**
- **#1318 gpio presence**

**17.07.12**
- #1941 BMC: Build: Create flash-able image with a static layout

- **#1842 BMC: Create Manifest file and add it to the <system>.tar image**
  * 7/17 - This will probably have to wait for 1941 above.

**17.07.26**

---

**Charlie: -**
**17.06.28**
- openbmc/openbmc-test-automation#746 Code update: Remaining upload image of BMC/ PNOR test cases
 
**17.07.12**
- **#1938 BMC: Reduce shutdown watchdog value back to default**

**17.07.26**

---

**Leo:lgon -**
**17.06.28**
- **#1550 Implement Software Object.Delete for pnor updater and image manager**

- #1663 Rename mboxd dbus object to be in correct xyz.openbmc_project

---

**Michael:mtritz -**
**17.06.28**
- **#1576 Implement BMC factory reset**

**17.07.12**
- **#1756 BMC: Implement the Activation flow (write/update files)**

**17.07.26**

---

**Saqib:saqibkh -**
**17.06.28**
- #1650 Move contents from /var/lib/obmc to /var/persist

**17.07.12**
- #1944 BMC: Block BMC reboot during u-boot update

**17.07.26**

---

gtmills, 2017-07-27:
Squad 2

Andrew:geissonator -

17.07.26:

#2032 (host reboot followed by chassis off results in reboot)
#1676 (reboot count decrement)
#1959 (vddr voltage setpoint increases)
#1958 (voltage power on workarounds)
#1709 (xcat inventory issues)
RAS error inject defect investigation
https://gerrit.openbmc-project.xyz/#/c/3749/ (old inventory removal)

Gunnar:gtmills:
17.06.28:

#1329 (power supply presence)
17.07.12:

#1860 (PDM elogs)
Chris:cbostic:

17.06.28:

#1783 (witherspoon gpio-keys)
17.07.12:

#1727 (ucd driver support)
Brandon:bjwyman: 80%

17.07.12:

#1835 (fan control final piece) -> Back to Barth
17.07.26:

#1731 (power supply monitor for faults)
Eddie:eddiejames: 80%

17.06.28:

#1883 (hwmon skeleton for power supply driver)
17.07.26:

BMC r/r
Button support (power on/off and reset)
Matt S:spinler: 80%
17.06.28:

#1724 (chip fault analysis)

17.07.26:

#1627 (systemd journal persistence and size restriction)
Matt B:msbarth: 80%
17.06.28:

#1835 (delay on events parameter)

#1839 (handle scale property)

Lower priority, finish as time allows
#1792 (yaml updates)

Focus on smart fan speed integration, need to ensure we don't destabilize witherspoon beta

Brad:bradbishop: 70%

17.05.31:

#1318 (gpio presence)
17.07.26:

Continue helping on code update squad
Dinesh:chinaridinesh: 65%

17.06.14:

#1315 (Ensure uboot fw_env update script uses new inventory object)

closed 2017-08-21 by zahrens