openbmc/openbmc #2001: [xCAT] How to clear event log?

xuweibj, 2017-07-21:
I saw that could get log information by using ``curl -k -b cjar https://bmc/xyz/openbmc_project/logging/enumerate``. 
Is there interface to clear log?

@whowutwut @bradbishop

mdmillerii, 2017-07-21:
https://github.com/openbmc/openbmc/issues/1561 
https://github.com/openbmc/openbmc/issues/1327

xuweibj, 2017-07-24:
Hi I have another question about event log. We got response like below:
```
"/xyz/openbmc_project/logging/entry/7": {
      "AdditionalData": [
        "TIMEOUT_IN_MSEC=2700000"
      ],
      "Id": 7,
      "Message": "xyz.openbmc_project.State.Host.Error.SoftOffTimeout",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1500649831845,
      "associations": []
    },
```
Will the format of ``Timestamp`` be changed in the future? Will it be like ``xx:xx:xx``?

whowutwut, 2017-08-10:
@dkodihal  Is there any plan here to create a README.md file to describe the use of the Logging API?

xuweibj, 2017-09-12:
Have got interface ``curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://x.x.x.x//xyz/openbmc_project/logging/action/deleteAll -d "{\"data\": [] }" ``. Close this issue.

closed 2017-09-12 by xuweibj