openbmc/openbmc #2002: [Barreleye] PNOR - Error 102 opening ffs !

swe12345, 2017-07-21:
PNOR - Error 102 opening ffs !

Driver info:
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-60"
VERSION_ID="v1.99.8-60-g4403eb0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-60"
BUILD_ID="v1.99.8"
```
```
root@barreleye:~# cat /etc/timestamp
20170720162802
```
```
root@barreleye:~# pflash -r /tmp/out.txt -P VERSION
Error 102 opening ffs !
```
```
root@barreleye:~# pflash -i
Flash info:
-----------
Name          = /dev/mtd6
Total size    = 64MB
Erase granule = 64KB

TOC@0x00000000 Partitions:
-----------
Error 102 opening ffs !
root@barreleye:~#
```
[journalctl_21jul.txt](https://github.com/openbmc/openbmc/files/1164996/journalctl_21jul.txt)

anoo1, 2017-09-12:
Could you try reflashing the pnor? Maybe it was blank or corrupted from a previous flash.

closed 2017-09-12 by anoo1

mdmillerii, 2017-09-13:
`libflash/libffs.h:#define FFS_ERR_BAD_CKSUM     102`

mdmillerii, 2017-09-21:
`#define FFS_ERR_BAD_CKSUM       102`