openbmc/openbmc #2003: dps310:  temp sensor or random number generator?

spinler, 2017-07-21:
We need to do some additional work.  From a string of emails:

" I might have an idea on why you are seeing this issue. We recently discovered that some older version samples had an issue where the gain bit for temperature needed to be manually set during power-up. 
  
 Bit 1 of address 0x32 must be 1. Do you read this from these samples? If that bit is 0, then do the following writes:   

write  0x0E to 0xA5
write  0x0F to 0x96
write  0x62 to 0x02
write  0x0E to 0x00
write  0x0F to 0x00

 This is a onetime configuration that you need to do as part of your startup routine. This should correct the temperature readings."

cbostic, 2017-07-24:
++++++++++++++++++++++++++++++++++++++
TASK LIST

- [ ] Identify workarounds

- [ ] Code up workarounds

- [ ] Verify correct temperature reading after loading fix

#**STATUS**

7/21
Coded up chip workaround to fix bad temperature readings
What's next: Test on a system with a dps310 available.   
Blockers: none

++++++++++++++++++++++++++++++++++++++

rfrandse, 2017-07-26:
<https://gerrit.openbmc-project.xyz/5528> kernel: ftgmac100, smc, dps310 fixes
Resolves:  #2003 dps310:  temp sensor or random number generator?

closed 2017-07-26 by williamspatrick