openbmc/openbmc #2004: UCD VCS rails failed to load [Witherspoon]

gkeishin, 2017-07-21:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-64"
VERSION_ID="v1.99.8-64-ga1fea2f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-64"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-d6ae40dc4c4dff3265cc254d404ed6b03fcc2206 #1 Fri Jul 21 14:21:07 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

```
```
Executing: GET  /xyz/openbmc_project/sensors/

/xyz/openbmc_project/sensors/
[
    "/xyz/openbmc_project/sensors/temperature",
    "/xyz/openbmc_project/sensors/fan_tach"
]
```

```
root@witherspoon:~#  ps | grep hwmon | grep -v grep
  968 root       0:24 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/max31785@52
  972 root       0:00 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@100/bmp280@77
 6556 root       0:01 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@70
 6561 root       0:00 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@140/ir35221@71
 6565 root       0:01 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@71
 6567 root       0:00 /usr/sbin/phosphor-hwmon-readd -o /ahb/apb/i2c@1e78a000/i2c-bus@180/ir35221@70
root@witherspoon:~#
```

```
Apr 20 18:06:36 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Apr 20 18:06:36 witherspoon kernel[923]: [ 2247.650000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Apr 20 18:06:37 witherspoon systemd[1]: Started Disable UCD VCS rails.
Apr 20 18:17:10 witherspoon systemd[1]: Stopped Disable UCD VCS rails.
Apr 20 18:17:21 witherspoon systemd[1]: Starting Disable UCD VCS rails...
Apr 20 18:17:23 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Apr 20 18:17:23 witherspoon kernel[923]: [ 2893.900000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Apr 20 18:17:23 witherspoon systemd[1]: Started Disable UCD VCS rails.

```

```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:~#
```

```
[  OK  ] Reached target Graphical Interface.
[  OK  ] Started Stop ureadahead data collection 45s after completed startup.
         Starting Update UTMP about System Runlevel Changes...
[  OK  ] Started Update UTMP about System Runlevel Changes.

Ubuntu Zesty Zapus (development branch) 05425 hvc0

05425 login:
Ubuntu Zesty Zapus (development branch) 05425 hvc0

05425 login:
```

```
root@witherspoon:/sys/class/hwmon/hwmon4# cat name
ucd90160
root@witherspoon:/sys/class/hwmon/hwmon4# ls
device            in11_max          in13_label        in15_crit         in16_max_alarm    in2_lcrit         in4_crit_alarm    in5_min           in7_lcrit_alarm   in9_input
in10_crit         in11_max_alarm    in13_lcrit        in15_crit_alarm   in16_min          in2_lcrit_alarm   in4_input         in5_min_alarm     in7_max           in9_label
in10_crit_alarm   in11_min          in13_lcrit_alarm  in15_input        in16_min_alarm    in2_max           in4_label         in6_crit          in7_max_alarm     in9_lcrit
in10_input        in11_min_alarm    in13_max          in15_label        in1_crit          in2_max_alarm     in4_lcrit         in6_crit_alarm    in7_min           in9_lcrit_alarm
in10_label        in12_crit         in13_max_alarm    in15_lcrit        in1_crit_alarm    in2_min           in4_lcrit_alarm   in6_input         in7_min_alarm     in9_max
in10_lcrit        in12_crit_alarm   in13_min          in15_lcrit_alarm  in1_input         in2_min_alarm     in4_max           in6_label         in8_crit          in9_max_alarm
in10_lcrit_alarm  in12_input        in13_min_alarm    in15_max          in1_label         in3_crit          in4_max_alarm     in6_lcrit         in8_crit_alarm    in9_min
in10_max          in12_label        in14_crit         in15_max_alarm    in1_lcrit         in3_crit_alarm    in4_min           in6_lcrit_alarm   in8_input         in9_min_alarm
in10_max_alarm    in12_lcrit        in14_crit_alarm   in15_min          in1_lcrit_alarm   in3_input         in4_min_alarm     in6_max           in8_label         name
in10_min          in12_lcrit_alarm  in14_input        in15_min_alarm    in1_max           in3_label         in5_crit          in6_max_alarm     in8_lcrit         of_node
in10_min_alarm    in12_max          in14_label        in16_crit         in1_max_alarm     in3_lcrit         in5_crit_alarm    in6_min           in8_lcrit_alarm   power
in11_crit         in12_max_alarm    in14_lcrit        in16_crit_alarm   in1_min           in3_lcrit_alarm   in5_input         in6_min_alarm     in8_max           subsystem
in11_crit_alarm   in12_min          in14_lcrit_alarm  in16_input        in1_min_alarm     in3_max           in5_label         in7_crit          in8_max_alarm     temp1_alarm
in11_input        in12_min_alarm    in14_max          in16_label        in2_crit          in3_max_alarm     in5_lcrit         in7_crit_alarm    in8_min           temp1_input
in11_label        in13_crit         in14_max_alarm    in16_lcrit        in2_crit_alarm    in3_min           in5_lcrit_alarm   in7_input         in8_min_alarm     uevent
in11_lcrit        in13_crit_alarm   in14_min          in16_lcrit_alarm  in2_input         in3_min_alarm     in5_max           in7_label         in9_crit
in11_lcrit_alarm  in13_input        in14_min_alarm    in16_max          in2_label         in4_crit          in5_max_alarm     in7_lcrit         in9_crit_alarm
root@witherspoon:/sys/class/hwmon/hwmon4#

```

- [x] Journald log  
[journald_v1.99.8-64-ga1fea2f_ucd_rail_failure.txt](https://github.com/openbmc/openbmc/files/1166226/journald_v1.99.8-64-ga1fea2f_ucd_rail_failure.txt)

gkeishin, 2017-07-21:
@spinler ^^^

spinler, 2017-08-03:
phosphor-hwmon doesn't find a configuration file for this chip, that's why it doesn't run.  It seems like maybe I removed the keywords from the MRW?  I can't quite remember.

I guess we should decide if we really want to expose the 16 voltages.  The UCD90160 itself is already monitoring these, so if one drops we'll get a power fault with the appropriate error logs.  Seems to me like it gives users warm fuzzies when they can check voltages themselves, but it would add at least 32 I2C ops per second to do so.

spinler, 2017-08-10:
We had a discussion today with the chief engineer, and he doesn't think customers would care about these voltages, so we're not going to add them to phoshpor-hwmon.  Closing.

closed 2017-08-10 by spinler