openbmc/openbmc #2006: fan watchdog expiration not affecting fans

spinler, 2017-07-21:
Summary:  when WDT1 expires, the fans don't change speeds, even though if I manually toggle that GPIO they will move to full.

Details:
I can see watchdog is happily pinging WDT1:
` 1105 root       0:00 /sbin/watchdog -T 3 -t 1 -F /dev/watchdog0`

And the WDT is indeed set up to toggle the GPIO on expiration:
```
# devmem 0x1e78500c
0x00000019   #enabled and gpio toggle
```
```
# devmem 0x1e785018
0x000000FF  #gpio pulse width
```

Kill fan control so watchdog dies and expires:
```
Jul 19 17:50:53 witherspoon systemd[1]: witherspoon-fan-watchdog.service: Killing process 1105 (watchdog) with signal SIGKILL.
Jul 19 17:50:53 witherspoon kernel[942]: [  370.180000] watchdog: watchdog0: watchdog did not stop!
```
watchdog stopped and expired:
```
# devmem 0x1e78500c
0x00000018
# devmem 0x1e785010
0x00000A01
```

Unfortunately, the fans were already at high, so hard to show them not go higher really, except that I am still able to manually change speeds down to 5000 RPM by writing the target, and that shouldn't be allowed if the watchdog had really taken over.

At this state, if I then manually toggle the WDT GPIO, then I can get fans going higher:
```
root@witherspoon:/sys/class/hwmon/hwmon1# cat *_input
4448
...
root@witherspoon:/sys/class/hwmon/hwmon1# phosphor-gpio-util -g 218 -p /dev/gpiochip0 -d 1 -a high_low
<wait some>
# cat *_input
10623
```
So I would say the path external to the AST is fine, the problem is with the actual GPIO toggle on watchdog expiration.  I had thought I had tested this a long time ago and it worked fine, well before any driver updates were made.  The pulse width is already set to the largest value possible - "256 timer base cycles".
  I may need to have the GPIO scoped to see what it is actually doing.

amboar, 2017-07-24:
Looking at the MAX31785 and AST2500 datasheets, the following note is made against `MFR_FAN_CONFIG`'s `TACHO` bit:

```
Note: If the fan fault is removed after ramping the PWM to 100% duty cycle, normal fan operation
is resumed.
```
The AST2500 datasheet says this about the `EXTRST#` pulse:

```
Reset width
This register decides the asserting duration of wdt_ext and wdt_rstarm signal. The default value is 0xFF. It means the default asserting duration of wdt_ext and _wdt_rstarm is 256us.
```
However the width of the field is 20 bits, and 0xFF only sets 8. It seems we could set a value of 0x1FFFFF to achieve a ~2s pulse? Regardless, when it deasserts it seems the fans just resume normal operation, hence your ability to change the fan rate.

Separately, we assert `MFR_FAN_CONFIG`'s `TSFO` bit, which disables the write watchdog and therefore the chip *doesn't* ramp the fans to 100% PWM duty for the period it doesn't receive writes to `FAN_COMMAND_1`. Deasserting `TSFO` might be a better approach, to allow the fans to be ramped across the time it takes for the system to reset, especially given it takes userspace around 90s to be fully functional (approximately 6 seconds of which is the kernel; I haven't measure the time spent in u-boot).

spinler, 2017-07-24:
Turns out there is a bit in WDT18 that says if the reset pulse is active high or active low, and it defaults to active low.  We need it to be active high, and also we need the open-drain vs push-pull bit to be push pull.  However even when I change those by hand I still haven't gotten it to work yet.  (Still works fine when I wiggle that GPIO by hand of course.)

spinler, 2017-07-24:
@amboar Could there be any pinctlr configuration that needs to be updated to use the WDT function instead of GPIO?

spinler, 2017-07-25:
# **STATUS**
* 7/24: still debugging

spinler, 2017-07-25:
Yea, it's the pinctlr config.  SCUA8 bit 2 needs to be set to enable WDT_RST1.  It's currently a 0.  Flipping it makes it work.

So that means 3 changes total:
1) pinctlr change to enable WDT_RST1
2) device driver change to WDT18[31] to set reset pulse polarity to open-drain active high
3) device driver change to WDT18[30] to set driving type to push-pull

I'm pretty bummed that I didn't realize that the WDT18 bits even existed before now or I would have asked for them when the other related watchdog driver changes were made.

amboar, 2017-07-25:
That should all be some straight forward devicetree changes, though maybe the bindings aren't defined for configuring the watchdog polarity. I'll check, and can send some patches.

spinler, 2017-07-27:
Assigned to @amboar as he is fixing.

spinler, 2017-08-01:
Fixes are coming in with https://gerrit.openbmc-project.xyz/#/c/5681/

geissonator, 2017-08-09:
Ready to close this one?  Looks like fixes are merged.

spinler, 2017-08-10:
Yea, fixed.

closed 2017-08-10 by spinler