openbmc/openbmc #2008: Interface and implementation of the boot policy setting

dkodihal, 2017-07-24:
IPMI has a concept of a one-time vs permanent boot settings. Investigate how this fits with new settings code and implement the same.

dkodihal, 2017-07-27:
- [ ] Add a boolean property to the Control/Boot/Mode interface to indicate that the boot process should preserve the specified Mode.
- [ ] Add a boolean property to the Control/Boot/Source interface to indicate that the boot process should preserve the specified Source.
- [ ] ipmid : read the relevant 'boot flag persist' property from the Mode setting object, or the Source setting object. Map 'true' to 'permanent' and 'false' to 'one-time'.

dkodihal, 2017-07-28:
- Figure out if the BMC has to consume the boot policy in a way other than just supplying it to a host. For example, should the BMC prevent preserving to the filesystem, a change to a boot mode/source setting, when the new setting is associated with a onetime policy?
- Find out how the host consumes the boot policy.

vishwabmc, 2017-07-28:
# **STATUS**

- 7/28:
   - Read the story and understood.

dkodihal, 2017-08-01:
This should remove unit/service dependencies on the old settings. That's still being retained now because of this single setting.

rfrandse, 2017-08-08:
<https://gerrit.openbmc-project.xyz/5840> Update the document with new settings information
Fixes  #2008 Interface and implementation of the boot policy setting

vishwabmc, 2017-08-10:
# **STATUS**

- 8/10:
  - Did search on ways to update password non interactive and found some examples. Need to test

rfrandse, 2017-09-01:
<https://gerrit.openbmc-project.xyz/6311> boot settings: use the new 'one time' setting
Resolves  #2008 Interface and implementation of the boot policy setting

closed 2017-09-08 by williamspatrick