openbmc/openbmc #2009: phosphor-network-manager: undefined symbol

shenki, 2017-07-24:
v1.99.8-71-ge319c8c

```
Jul 24 04:08:43 witherspoon systemd[1]: Starting Phosphor BIOS Code Update...
Jul 24 04:08:43 witherspoon systemd[1]: Starting Phosphor BMC Code Update...
Jul 24 04:08:44 witherspoon phosphor-network-manager[1028]: /usr/sbin/phosphor-network-manager: symbol lookup error: /usr/sbin/phosphor-network-manager: undefined symbol: _ZTVN9sdbusplus3xyz15openbmc_project7Network6server10MACAddressE
Jul 24 04:08:44 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Main process exited, code=exited, status=127/n/a
Jul 24 04:08:44 witherspoon systemd[1]: Failed to start Phosphor Network Manager.
Jul 24 04:08:44 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Unit entered failed state.
Jul 24 04:08:44 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Failed with result 'exit-code'.
Jul 24 04:08:44 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Service hold-off time over, scheduling restart.
Jul 24 04:08:44 witherspoon systemd[1]: Stopped Phosphor Network Manager.
Jul 24 04:08:44 witherspoon systemd[1]: Starting Phosphor Network Manager...
Jul 24 04:08:44 witherspoon phosphor-network-manager[1037]: /usr/sbin/phosphor-network-manager: symbol lookup error: /usr/sbin/phosphor-network-manager: undefined symbol: _ZTVN9sdbusplus3xyz15openbmc_project7Network6server10MACAddressE
Jul 24 04:08:44 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Main process exited, code=exited, status=127/n/a
Jul 24 04:08:45 witherspoon systemd[1]: Failed to start Phosphor Network Manager.
Jul 24 04:08:45 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Unit entered failed state.
Jul 24 04:08:45 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Failed with result 'exit-code'.
Jul 24 04:08:45 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Service hold-off time over, scheduling restart.
Jul 24 04:08:45 witherspoon systemd[1]: Stopped Phosphor Network Manager.
Jul 24 04:08:45 witherspoon systemd[1]: Starting Phosphor Network Manager...
Jul 24 04:08:45 witherspoon phosphor-network-manager[1041]: /usr/sbin/phosphor-network-manager: symbol lookup error: /usr/sbin/phosphor-network-manager: undefined symbol: _ZTVN9sdbusplus3xyz15openbmc_project7Network6server10MACAddressE
Jul 24 04:08:45 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Main process exited, code=exited, status=127/n/a
Jul 24 04:08:45 witherspoon systemd[1]: Failed to start Phosphor Network Manager.
Jul 24 04:08:45 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Unit entered failed state.
Jul 24 04:08:45 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Failed with result 'exit-code'.
Jul 24 04:08:46 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/led/groups/bmc_booted.
Jul 24 04:08:46 witherspoon systemd[1]: Started Wait for /org/openbmc/NetworkManager/Interface.
Jul 24 04:08:46 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/led/groups.
Jul 24 04:08:46 witherspoon systemd[1]: Started Phosphor BIOS Code Update.
Jul 24 04:08:46 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Service hold-off time over, scheduling restart.
Jul 24 04:08:46 witherspoon systemd[1]: Stopped Phosphor Network Manager.
Jul 24 04:08:46 witherspoon systemd[1]: Starting Phosphor Network Manager...
Jul 24 04:08:46 witherspoon systemd[1]: Started FRU Fault monitor service.
Jul 24 04:08:46 witherspoon phosphor-network-manager[1047]: /usr/sbin/phosphor-network-manager: symbol lookup error: /usr/sbin/phosphor-network-manager: undefined symbol: _ZTVN9sdbusplus3xyz15openbmc_project7Network6server10MACAddressE
Jul 24 04:08:47 witherspoon systemd[1]: Started Phosphor DNS Settings Cache Manager.
Jul 24 04:08:47 witherspoon systemd[1]: Starting Assert bmc_booted LED...
Jul 24 04:08:47 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Main process exited, code=exited, status=127/n/a
Jul 24 04:08:47 witherspoon systemd[1]: Failed to start Phosphor Network Manager.
Jul 24 04:08:47 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Unit entered failed state.
Jul 24 04:08:47 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Failed with result 'exit-code'.
Jul 24 04:08:47 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Service hold-off time over, scheduling restart.
Jul 24 04:08:47 witherspoon systemd[1]: Stopped Phosphor Network Manager.
Jul 24 04:08:47 witherspoon systemd[1]: Starting Phosphor Network Manager...
Jul 24 04:08:47 witherspoon phosphor-network-manager[1052]: /usr/sbin/phosphor-network-manager: symbol lookup error: /usr/sbin/phosphor-network-manager: undefined symbol: _ZTVN9sdbusplus3xyz15openbmc_project7Network6server10MACAddressE
Jul 24 04:08:47 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Main process exited, code=exited, status=127/n/a
Jul 24 04:08:47 witherspoon systemd[1]: Failed to start Phosphor Network Manager.
Jul 24 04:08:47 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Unit entered failed state.
Jul 24 04:08:47 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Failed with result 'exit-code'.
Jul 24 04:08:47 witherspoon systemd[1]: Started Assert bmc_booted LED.
Jul 24 04:08:47 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Service hold-off time over, scheduling restart.
Jul 24 04:08:48 witherspoon systemd[1]: Stopped Phosphor Network Manager.
Jul 24 04:08:48 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Start request repeated too quickly.
Jul 24 04:08:48 witherspoon systemd[1]: Failed to start Phosphor Network Manager.
Jul 24 04:08:48 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Unit entered failed state.
Jul 24 04:08:48 witherspoon systemd[1]: xyz.openbmc_project.Network.service: Failed with result 'exit-code'.
```

geissonator, 2017-07-24:
Probably related to #2011

ratagupt, 2017-07-24:
No,it should not be related to 2011.

shenki, 2017-07-25:
I don't see it on v1.99.8-79-g246c6f9 either.

closed 2017-07-25 by shenki